openshwprojects/OpenBK7231T_App #20: OTA changes all devices name to obk8C000000

boozeman, 2022-02-13:
Testing OTA from device.ip.address/app page. All went well, but restore did not work, the device drops at AP-mode at the process and MAC address somehow changed to C8:47:8C:00:00:00 and the device name is obk8C000000 1. 
I was testing this with 2 devices and both have the same behavior. Original device names was obk61FB56C0 1 and obk618184AB 1

Firmware was build in current git state.

My router (and anyone elses too) did not like 2 devices with the same mac address. 

My humble suggestion is that we can later on use device name (and possibly the mqtt-topic name) as the same as configured MQTT settings client name.

openshwprojects, 2022-02-16:
The current version of firmware has added mac change functionality (for both Bekken and XR809 chips). Is it working for you now?

boozeman, 2022-02-18:
The problem seems to be with only these two modules whose firmware was updated incorrectly by me. If any correction is possible I would be grateful. I can always try change the modules for the devices because I have two extra ones.

openshwprojects, 2022-02-18:
It will be possible, we just need to look into it, but's its not top priority..

openshwprojects, 2022-02-19:
I think we can fix that, just reconstruct TLV header

boozeman, 2022-02-20:
If I read RF Config for this wrecked module, I got table full of FF. 0000 to 0FE0. Working ones have MAC address at 0020.