openshwprojects/OpenBK7231T_App #63: volatile reference

jbanaszczyk, 2022-03-20:
`new_pins.c` many occurrences of `volatile UINT32 *` : are you sure that it was not `UINT32 * volatile` ?

btsimonh, 2022-03-20:
e.g.

	volatile UINT32 *gpio_cfg_addr;
	gpio_cfg_addr = (volatile UINT32 *)(REG_GPIO_CFG_BASE_ADDR + index * 4);

the variable which could be changed, and so not cached is the content of the pointer, not the pointer itself.

referring to https://barrgroup.com/embedded-systems/how-to/c-volatile-keyword#:~:text=C's%20volatile%20keyword%20is%20a,code%20the%20compiler%20finds%20nearby.

it looks correct to me?

btsimonh, 2022-03-20:
closing as I believe it is correct :).
thanks for highlighting that it should be checked!

closed 2022-03-20 by btsimonh