openshwprojects/OpenBK7231T_App #64: Local nonstatic variables are not initialized to 0

jbanaszczyk, 2022-03-20:
https://github.com/openshwprojects/OpenBK7231T_App/blob/e109dce22f497413f28c199064a7d22b46fe749b/src/new_common.c#L55

`a` and `b` contains garbage

btsimonh, 2022-03-20:
they are initialised (in a strange way!) before use:
        int a, b;
	// avoid signing issues in conversion to int for isxdigit(int c)
	const unsigned char *src = (const unsigned char *)srcin;
        while (*src) {
				if(curLen>=maxDstLen)
					break;
                if ((*src == '%') &&
                    ((a = src[1]) && (b = src[2])) &&

but agree...  in case code below is modified, we should init them.
The compiler does produce warnings if there is a chance of a variable being used before initialisation.

btsimonh, 2022-03-20:
pls review PR

closed 2022-03-20 by btsimonh