openshwprojects/OpenBK7231T_App #66: Memory leaks

jbanaszczyk, 2022-03-20:
https://github.com/openshwprojects/OpenBK7231T_App/blob/e109dce22f497413f28c199064a7d22b46fe749b/src/new_common.c#L55

2x `malloc` without `free`

btsimonh, 2022-03-20:
test1 and test2?
the can be removed now - I was concerned at one point that the allocated table was being overwritten, so these are just a couple of 'guard' areas containing known data, HOPEFULLY either side of the *initial* cfg area (since it's allocated close to startup, but proximity is in no way guaranteed - so they are actually of little use!).
But in any case, they should only be allocated once at startup, and never again.

btsimonh, 2022-03-20:
p.s. pls take a look at 
https://github.com/openshwprojects/OpenBK7231T_App/pull/67
and confirm it addresses the issues you have highlighted.

also, PRs welcome :).

I'll close the issues for now, re-open if necessary?

closed 2022-03-20 by btsimonh