openshwprojects/OpenBK7231T_App #68: Tuya MCU Implementation

axcs, 2022-03-22:
I can can implement on configurations, Tuya MCU on pinout configurations?

openshwprojects, 2022-03-22:
Not yet, but TuyaMCU will be soon configurable by simple text config file. Text file will allow people to specify which dpId maps to which channel and which function. This will allow to add support to many TuyaMCU devices without recompiling the source. Please wait some time and watch our repo, the feature will surely show up soon.

axcs, 2022-03-27:
please when there is news, or it is necessary to test something please let me know

btsimonh, 2022-04-06:
@axcs - I'm currently re-working TuyaMCU in TAS, and it's a pig.
What device(s) do you have (i.e. what support are we looking for first)?

openshwprojects, 2022-04-27:
You should see this:
https://github.com/openshwprojects/OpenBK7231T_App/pull/75
I have tested it and I confirm that dimmer with TuyaMCU works.

openshwprojects, 2022-06-01:
TuyaMCu update https://www.elektroda.com/rtvforum/viewtopic.php?p=20031489#20031489

openshwprojects, 2022-06-10:
Dimmer guide: https://www.elektroda.com/rtvforum/topic3898502.html
Fan guide: https://www.elektroda.com/rtvforum/topic3895301.html

closed 2022-06-10 by openshwprojects