openshwprojects/OpenBK7231T_App #72: bk7231N socked

JanKiel2000, 2022-04-12:
Hello.
Can you get on the addition of Volt, AMPER, WAT reading. I have two sockets with a measurement and I would like to check.
Do you think to add MQTT support for Domoticz?

openshwprojects, 2022-04-13:
What kind of chip is used for power measurement in your sockets? What are the socket models?
So far I have only BL0942 support, but there are some similiar chips, so more to come soon. I will publish a guide about power metering soon.

What is required for "Domoticz MQTT support"?

JanKiel2000, 2022-04-13:
The alley is on the BK7231N chip. I have already uploaded your soft and works. Of course without measuring electricity. The measurement chip is BL0937.
When it comes to domotico support.
The MQTT device sends "domoicz/in"
Domoticz corresponds to "domotiz/out"
Controllers are identified after IDX.

domoticz/in = {"idx":78,"nvalue":1,"svalue":"","Battery":100,"RSSI":8} - nvalue 1-on , 0 - off
domoticz/in = {"idx":88,"nvalue":0,"svalue":"215","Battery":100,"RSSI":8} - svalue is Voltage
domoticz/in = {"idx":89,"nvalue":0,"svalue":"0.213","Battery":100,"RSSI":8} - svalue is Ampere
domoticz/in = {"idx":102,"nvalue":0,"svalue":"45;15318.6","Battery":100,"RSSI":8} - svalue is Wat, WatHour
domoticz/in = {"idx":91,"nvalue":0,"svalue":"15318.6","Battery":100,"RSSI":8} - svalue is WatHour
RSSI from 1 to 12, 

domoticz/out = {"Battery" : 255, "LastUpdate" : "2022-04-13 19:33:39", "RSSI" : 12,	"description" : "", "dtype" : "Light/Switch", "hwid" : "7",	"id" : "0001409E","idx" : 78,	"name" : "Kodi sypialnia", "nvalue" : 0,"stype" : "Switch",	"svalue1" : "100", "switchType" : "On/Off", "unit" : 1} where nvalue = 1 -on, 0 - off

openshwprojects, 2022-04-18:
And you set "idx" value by hand?
Can you help testing it when I implement that?

What in case of relay controller with 2 relays, how it corresponds to to nvalue?

Why is there a "battery" field in domoticz/out string, when it is send out by domoticz?

JanKiel2000, 2022-04-18:
You can read "IDX" in Domoticz and enter it into the device.  Each relay has its own "IDX".  So two relays are two "IDX".  Parameters such as "Battery", "RSSI" are optional and can be dealt with later or not at all.  They look nice in the table.  To control a relay, it is really enough to enter its "IDX" and "nvalue".  0 is off, 1 is on.  Other elements in Domoticz to which, for example, voltage is sent, also have their "IDX" but their value is already sent in "svalue".  "nvalue" is then disregarded.  Will be very happy to test.  I have two sockets with this chip and I hope to integrate them with my network and Domoticz.  I still use AFE and Tasmota in ESP devices.

JanKiel2000, 2022-05-03:
Any progress?

openshwprojects, 2022-06-01:
basic BL0397 support has been added, but timer mechanism will be changed to more precise soon, @JanKiel2000

JanKiel2000, 2022-06-01:
Unfortunately, it doesn't work for me. The "Startdriver Bl0937" command shows zero on your FV.
I discovered that I have other GPIO
`int GPIO_NRG_SEL = 24; // pwm4
int GPIO_HLW_CF = 7;
int GPIO_NRG_CF1 = 6;`
After compiling the code with my settings, unfortunately, the device is reset. As I introduce a request to this type of code
`void HlwCf1Interrupt(unsigned char pinNum) {  // Service Voltage and Current
     //g_vc_pulses++;
}

void HlwCfInterrupt(unsigned char pinNum) {  // Service Power
   // g_p_pulses++;
}
`
This device does not restart but also shows nothing (which is obvious).

ec-blaster, 2022-06-03:
I have a similar problem with BL0937.
The driver's default GPIO's interfere with one of my device's LED.
It would be ideal if we could **configure the custom GPIO's of the driver in each device**. I thought we could add some additional commands to set them at run time, but then I realized that if you issue an additional command, the driver is already connected (or kind of).

I'll try to compile my own firmware tonight

ec-blaster, 2022-06-03:
I was able to compile it with my changes. I opened a new issue here with the results: #92

closed 2022-11-04 by openshwprojects