openshwprojects/OpenBK7231T_App #76: Flashing on WB3S not working!

LuisThe0ne, 2022-04-24:
Hey Guys! 

I have a problem, whenever I try to flash the WB3S with OpenBeken it simply doesn't work.
I wired it like this:
[image]
(I don't have an usb uart adapter thats why i am using an Arduino, for tasmota that setup worked well)

If I start bkwriter it starts flashing but never goes further than Init and after a while it says failed.(I tried the python program too -> Cannot get bus)

Has anyone had the same problem and knows a solution?

Thanks

PS: Yes I tried with resetting after starting bkwriter and without.

openshwprojects, 2022-04-24:
Hello, I will try to help you, altough I would note here that this issue has nothing to do with OpenBeken, it's just flashing issue you have.
So, I can see multiple problems:
1. RX goes to TX and TX to RX and not the other way 'round
2. RX and TX should have 3.3V logic levels (as I wrote on Elektroda guides) and Arduino most likely have 5V....
3. After rebooting, the electrolytic caps from board takes too much current and might reset USB to UART bridge (solution: for example, power from separate power supply or port)
4. Are you sure that there is nothing on RX and TX lines, like, for example, a button pull up resistor or a switch or anything? In that case you need to remove that part from TX/RX line, or cut a trace, etc....

https://www.elektroda.com/rtvforum/topic3875654.html

zalmyr, 2022-04-27:
I had to use the uart command line tool to flash the WB3S

openshwprojects, 2022-04-27:
> I had to use the uart command line tool to flash the WB3S

it's strange, I am programming a lot of devices with WB3S and WB2S and BKwriter 1.60 always worked for me (but sometimes the lower baud doesn't work, the second lowest is ok).

LuisThe0ne, 2022-05-01:
First of all,
thanks for your help!
I think I accidentely destroyed my Wb3s before I could try your Ideas (I applyed 5v to the 3.3v and now it sends infinite 00 to the serial port), but if I manage to get it working  I will let you know if the steps worked!

Regards Luis

closed 2022-06-19 by LuisThe0ne