openshwprojects/OpenBK7231T_App #78: OTA Bricking?

brandondb1, 2022-04-26:
Hi, I flashed .016 successfully using the Python tool.  When I tried to OTA upgrade using the web interface (not the app) I used the UG variant - is that correct?  Or use the java app and the .rbl?

Also the URL I used was file:///c:/directory/to/bin (I don't have a web server running right now).

This should work, but in any case it seems like I've now bricked the device as the web interface won't come up, nor does the AP come up.  When I try to reprogram it using the Python tool it won't find the bus (I'm not sure if you can re-program with it or not).

This is for the 7231N chip.

In any case, I'm sure I did something incorrectly but it might be useful to trap improper actions in code as it was really easy to break it (saying this as a veteran of many tasmota OTA upgrades).  I can help try and recreate the conditions if that helps.

Thanks for your great work on this!

openshwprojects, 2022-04-27:
For OTA, you should always use RBL.
I will add a notice about that on the OTA pages.

In the current state of things, you should always use java panel, it's easier to drag and drop file.

Also...  the native interface (not the app) works by sending HTTP GET so I think it's impossible for it to get a file with no running web server.

You can always unbrick device, even if the UART flasher is not working, you can always use SPI method. 
https://github.com/OpenBekenIOT/hid_download_py/blob/master/SPIFlash.md

Dobie007, 2022-05-17:
Hi, 
I am using a web server on my Linux PC by using this command out of the directory with the rbl file in it: " python3 -m http.server"  This is what i get as response:
Serving HTTP on 0.0.0.0 port 8000 (http://0.0.0.0:8000/) ...

When I open firefox with "0.0.0.0:8000" I can see the contents of the directory. 
From a different pc in the network i can see it with "http://[ip of the host]:8000" as well.
[image]

I copied the link with the host ip to the rbl file from there and used the OTA update function. 
On the linux pc i see this in the terminal: 
192.168.0.204 - - [17/May/2022 19:53:46] "GET /OpenBK7231T_1.0.20.rbl HTTP/1.1" 200
So the Beken chip is requesting the file as it should...

After that i waited for a minute or so and then did a network scan using arp - and got no response. 
The Beken Chip is now showing up as access point again...  hopefully with the new firmware.

openshwprojects, 2022-05-18:
@Dobie007 I think I should clarify few things
1. The method you shown might be useful only for older versions, because they had low URL lenght limit and it was not possible to use github link directly:
http://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.0.20/OpenBK7231N_1.0.20.rbl
2. For newer versions, as said above, use direct github link
3. "showing up as access point again" - the very old versions have not compatible config format and it will reset to open access point, but newer versions should always keep config after upgrading

Dobie007, 2022-05-18:
Ok thanks. Then my previous version was too old because I actually tried updating directly from Github without success.

closed 2022-06-10 by openshwprojects