openshwprojects/OpenBK7231T_App #79: Home assistant auto setup

brandondb1, 2022-04-26:
Hello,

As you know the MQTT built into Home Assistant has pretty strong auto setup abilities.  Although Tasmota has gone away from this to their own integration, I have maintained a custom version of Tasmota meant to work with the MQTT integration.

Although I am not a great programmer by any stretch, I understand the Home Assistant MQTT auto setup quite well and could help if someone wanted to try and extend the MQTT in that direction.

I also saw that someone was asking about Domoticz - there is a Home Assistant MQTT plugin for Domoticz that uses the same autosetup protocols (and one for OpenHab too), so working on the auto setup for Home Assistant MQTT could provide an easy path to setting up in all three systems.

LuisThe0ne, 2022-06-19:
It would also be an option, to build a home assistant integration, like tasmota did that recognizes ObenBeken devices.

TurkeyMan, 2022-06-27:
Is there an advantage to a custom integration? Wouldn't it be better to work by default in the way the OP suggests?

btsimonh, 2022-06-27:
One way I considered doing this was to store the template on the device, but construct it in the webapp.  In this way, the webapp can be quite sophisticated in the way the template is constructed, without loads of hard coded stuff in the firmware. (I did some TAS MQTT stuff in the past - BLE, for example - and it seems pretty nasty to have hand coded up json in the source code....).
The devices (not sure about all processors) have a filesystem available, within which we can store generic files.  So to serve them over the MQTT should be very easy, and the webapp can be used to create specific HA templates with much easier to program Javascript.

belikh, 2022-07-05:
If you're using MQTT anyway can't you just store things like this in a retained message?

Mordevolt, 2022-07-08:
is it possible to use Home Assistant MQTT Discovery? as per documentation it is probably pretty simple: https://www.home-assistant.io/docs/mqtt/discovery/

brandondb1, 2022-07-08:
It is quite simple to use the Hass matt autosetup...just need to publish a JSON with the right info in it, to the right topic.

iprak, 2022-09-03:
@openshwprojects I am trying to implement this but cannot locate cJSON implementation in Tuya SDK. There are a bunch of header files under `sdk\include` but no implementation?

openshwprojects, 2022-10-02:
@iprak basically nailed it, good job, discovery seems to be working.

https://github.com/openshwprojects/OpenBK7231T_App/pull/183

More to come soon.

closed 2022-10-02 by openshwprojects