openshwprojects/OpenBK7231T_App #84: Source of embedded toolchain

owinebar, 2022-05-17:
For the BK7231t variant, the original build_app.sh will download a toolchain from the URL:
https://airtake-public-data-1254153901.cos.ap-shanghai.myqcloud.com/smart/embed/pruduct/bk7231t_1.1.2.zip
At least some of the components of that zip file appear to be from an old GNU ARM embedded toolchain source.  Other parts are chip specific.  The file "bk7231t.yaml" indicates the files are provided under an MIT license, and that the source  should be available from:
    git: https://code.registry.wgine.com/hardware_wifi_soc/bk7231t.git
    tag: 1.0.1
However, I cannot reach that URL, so I cannot reproduce the GNU parts of the toolchain from source to verify it is a vanilla build.  
Since you have included the toolchain in your repo, the build process should not fetch the toolchain from the URL above, but I don't know whether the toolchain you have included is from that URL or a more authoritative source, e.g. based on https://developer.arm.com/Tools%20and%20Software/GNU%20Toolchain.
Has anyone tried using standard releases of the embedded toolchain in place of Tuya's custom archive?  Or replacing the other binaries in the zip file from standard sources (or producing equivalents)?

openshwprojects, 2022-05-17:
We didn't download any of the toolchains, the toolchain you mention is already downloaded in the original tuya-iotos-embeded-sdk-wifi-ble-bk7231t, same goes for tuya-iotos-embeded-sdk-wifi-ble-bk7231n.

The toolchain git link is not working for me.

EchterAgo, 2022-06-27:
All the toolchain binaries in `bk7231n_0.0.21.zip`, except some CRT object files that are not present, are the same as in the toolchain [gcc-arm-none-eabi-4_9-2015q1-20150306-linux.tar.bz2](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update/+download/gcc-arm-none-eabi-4_9-2015q1-20150306-linux.tar.bz2) from https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update which [includes the source code](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update/+download/gcc-arm-none-eabi-4_9-2015q1-20150306-src.tar.bz2).

closed 2022-11-04 by openshwprojects