openshwprojects/OpenBK7231T_App #85: MQTT connection problem

iprak, 2022-05-17:
I recently flashed a couple BK7231T WB2S devices (same devices) with the latest OpenBK7231T_UA_1.0.20.bin image. I had a hard time getting BKwriter to work but was able to make it work in the first attempt using hid_download_py.

The problem I am running into is with MQTT connection. I ahve noticed that if I leave user/password empty then MQTT is 0 in the log but if I enter something random e.g. user/password then connection succeeds. My MQTT server does not require authentication. Any suggestions for further debugging?

openshwprojects, 2022-05-17:
This is something new, I have never tested it with MQTT server with no auth required.

Can you check how the send packets looks like in Wireshark?

The original MQTT library did not support auth, but I added a support for that:
https://github.com/openshwprojects/OpenBK7231N/commit/172f1012c3a03b73441abc24a9f8ddd11e2923f1
that was required for both N and T platforms, but right now I can't see anything wrong in that code.

alex3594, 2022-05-26:
FYI, I checked that with tcpdump and can confirm when no user and password set, device is not sending any packets to the MQTT.

openshwprojects, 2022-06-05:
should be fixed
https://github.com/openshwprojects/OpenBK7231T/commit/2c3af1f719cb5d52f3a9442bc0ec4439cac1d389

closed 2022-06-08 by openshwprojects