openshwprojects/OpenBK7231T_App #86: 63-character wifi password key not working

Weltkugel, 2022-05-21:
Hi,
I've tried using a 63-character wifi password key, but only shorter passwords (up to 62-character) are working. Would you be so nice an fix this minimal issue? Thank you.

PS: Thank you for this great possiblity to use the CB2S with cloudfree software

openshwprojects, 2022-05-21:
Well , it seems you are correct. There was an error in the safe strcpy routine which was being a bit too safe and leaving one more byte too early. 
It's fixed now in 6f58f0631af16d936c3001d84689d834816db681
Please recompile and recheck, or wait for next release.

PS: Why do you need 63 letters wifi pass? Really?

Weltkugel, 2022-05-21:
Thank you. That was fast. Yeah some people are really paranoid ;-)
To be honest, it's the same 16 character password 4 times in a row (without the last character). I found this 63-character issue in different projects (a lot of china wlan-cameras understand only 31 character) .

closed 2022-05-21 by Weltkugel