openshwprojects/OpenBK7231T_App #87: Autostart doesn't work

alex3594, 2022-05-25:
Hello!
Is there any way to switch on the light for the Tuya E27 LED bulb?
I tried to add "color #000000FF" to the "startup command line", but it didn't help. 
Thanks!

openshwprojects, 2022-05-25:
Really? Let me check... RGBCW Tuya bulb here. E27.

[image]

after reboot:
[image]

well, it works for me. Remember thatyou can also use new led driver commands as well.

alex3594, 2022-05-25:
Sorry, I meant I need to make it light right after the startup (when you powered the device up). 
All controls work fine except autostart the light.

openshwprojects, 2022-05-25:
Really? That's interesting, let me check....

ah, now I see what you mean! It's not a bug, it's a feature.
If you do more than 2 reboots without waiting for 30 seconds, then device assumes that it has a technical problem and enters "safe mode" so youi can recover it from bad config.

So it works even if you unplug, but only if you wait 30 seconds after reach reboot.

alex3594, 2022-05-25:
Thanks for the clarification! It works fine after I wait for 30 seconds!

alex3594, 2022-05-31:
It will be great if you add something like "5 restarts in 10 seconds" for the recovery mode if possible.
Thanks!

openshwprojects, 2022-06-06:
@alex3594 
i have changed it and now you need at least 5 boot failures to run safe mode (but still full boot counts as 30 sec)
https://github.com/openshwprojects/OpenBK7231T_App/commit/65210cd44b44e8a7ffa112f5cd3990e7e9f848de
will it be enough?

alex3594, 2022-06-06:
@openshwprojects 
Thanks for the update, but I think I need this timeout to be removed (or make it 1 second) if possible. 
Because in my scenario I may need to power off the light, then power it on again in a few seconds. 
But I'm pretty sure it will not be powered off on on 5 times.

openshwprojects, 2022-06-06:
@alex3594 
What about this?
https://github.com/openshwprojects/OpenBK7231T_App/commit/35470c431c71e7e8babbcdbe51517907b02bdbff
This will let you to set time required for full boot (at least 1 second)

but use wisely, it will be very easy to lock yourself out with such a short boot ok delay

not fully tested

Time is changed in Config->General

openshwprojects, 2022-07-01:
Well, you haven't responded in a long time, and requested feature is implemented, so I will assume that's it resolved.

closed 2022-07-01 by openshwprojects