openshwprojects/OpenBK7231T_App #90: T34 compatibility

ec-blaster, 2022-05-30:
Hello.

I've been reading about your effort to flash a custom firmware in these devices and I find it amazing.
I have a new Tuya wifi in-wall socket with power meter which has serious wifi stability problems, I opened it and I can see there's a "T34" chip inside. I've been reading the characteristics and it seems to be based on BK7231N (according to this [Tuya documentation](https://developer.tuya.com/en/docs/iot/t34-module-datasheet?id=Ka0l4h5zvg6j8)).

[image]

I wonder if your project might be compatible with this chip.

Thanks.

openshwprojects, 2022-05-31:
It is most likely already compatible and if not, then I can try adding it, if that's needed.

The only problem is I don't have any T34 chips at hand and I have never seen such a module before.

Have you tried flashing one?

ec-blaster, 2022-05-31:
I've just flashed following your guide and the Tuya pinout documentation and it seems to work. The firmware flashes OK and I can configure the device. Now I'll try to get the GPIOS working. I'll let you know, I have some photos too.

ec-blaster, 2022-06-06:
I'll close the issue, as T34 has proven to be compatible.
Thanks.

closed 2022-06-06 by ec-blaster