openshwprojects/OpenBK7231T_App #91: Adding support for IR Reciever/Transmitter

rgtkd, 2022-05-30:
Hello, would be possible to add support or configure an IR Transmitter/Receiver?

Bought an S06 Generic transmitter and has a CBU board (BK7231N). Managed to flash OpenBk on it and works great, I can control an LED and button that the board includes, but have no way to configure the transmitter/receiver that Im aware of.

This is very close to the model I bought but instead of ESP8266 it comes with the CBU board:
https://templates.blakadder.com/auvisio_S06.html

I have some Ardunio/Programming experience, so if there is anywere documentation on how to add custom input/outputs I could try to implement this, however I am not able to find any kind of documentation or way on how to do this.

I would be able to test and try to help with the implementation if possible, it will be great if we can get this transmitters working!

Regards.

Sky-Jumper, 2022-06-30:
Hello!
Tasmota uses the [IRremoteESP8266 library](https://github.com/crankyoldgit/IRremoteESP8266) that supports numerous protocols. Can it be used somehow? It would be nice.

rgtkd, 2022-06-30:
> We've been discussing. We have a couple of issues, c++ support being one, and the other being a high res timer. Of course the C++ thing is just getting the makefiles right (help welcome!). The hi-res timer is more problematic, especially across multiple processors. For Beken I spent 2 days looking, and with so little info on the processor, failed to find a good solution for IR input.
> 
> We've mainly focussed on IR _in_ previously. For the OP, we'd need (I guess?) 38khz modulation too. So it's like we'd use PWM configured to produce the modulation, and a separate ~560us interrupt to control it (to post bits out). Since I would also like to control multiple IR devices, I may just have to buy the unit he mentions!

https://es.aliexpress.com/item/4000825628583.html

This is the one I bought, can provide help with any testing if needed thou, I currently have the chip desoldered, so can do anything you need with it.

Regards

avdmr, 2022-07-15:
I would also really like this. I have the s08 but the insides of it seem to be the exact same as the s06. 
Here is a link to the version I bought for reference: https://www.aliexpress.com/item/4000719809991.html
I would also be happy to help testing.

kevinblo, 2022-07-17:
Is there any hope that you will implement this feature?

openshwprojects, 2022-07-30:
@btsimonh any news on this?

rlsvr, 2022-09-16:
I am also interested in this topic, I have an IR remote with cb3s chip. [This.](https://www.moeshouse.com/collections/smart-link-remote/products/smart-infrared-rf-wifi-remote-controller-wireless-usb-ir-controller)
[image]
[image]

btsimonh, 2022-10-19:
For those following this topic, it's almost ready :).

openshwprojects, 2022-11-04:
IRReceiver works great on N and T. Special thanks for @btsimonh  for great input into the development and constant support and advising. It's time to grab remotes!

closed 2022-11-04 by openshwprojects