openshwprojects/OpenBK7231T_App #92: BL0937 problem

ec-blaster, 2022-06-03:
Hi there.

I have [this Tuya socket](https://www.aliexpress.com/item/1005001412312307.html?spm=a2g0o.order_list.0.0.191818022zImOh) with power meter. It uses the new "[T34](https://developer.tuya.com/en/docs/iot/t34-module-datasheet?id=Ka0l4h5zvg6j8)" chip, which is based on **BK7231N**. The power meter chip is a **BL0937**. I talked about this before: #90.

Following your instructions and reading several posts in elektroda.com, this is what I achieved until now:

- I flashed your BK7231N firmware and the basic socket functionality is working (ON/OFF and leds). These are the pins I used:
    | GPIO | Function | Channel | Comments |
    | ------|----------|----------|------------|
    | P7     | REL_N    | 1            | Main Relay |
    | P8     | WIFI LED_N | 0      | Blue light, blinking when connecting to wifi, OFF when connected |
    | P9     | BTN       | 1            | Button that toggles relay when pressed |
    | P23   | LED_N   | 1            | Red light, ON when relay is on |
- I tried to start the BL0937 driver executing the custom command `startDriver BL0937`, but the measures are zero and it interferes with my settings, as it is using GPIOS 7 and 8 by default.
- With a multimeter, I was able to check which GPIOs the device is using to connect to the BL0937. These are my results:
    | Device legend | T34 pin | T34 GPIO | BL0937 pin |
    |---------------|---------|------------|------------|
    | H                    | 20         | P20          | SEL            |
    | CF                   | 18         | P22          | CF            |
    | CV                  | 16         | P24          | CF1            |
- So I tried to modify the source code of the driver (`drv_bl0937.c`), making the following changes:
   ~~~ c
   ...
   #define ELE_HW_TIME 1
   #define HW_TIMER_ID 0

   // Original firmware GPIOS
   //int GPIO_NRG_SEL = 24; // pwm4
   //int GPIO_HLW_CF = 7;
   //int GPIO_NRG_CF1 = 8;

   // My device GPIOS
   int GPIO_NRG_SEL = 20;
   int GPIO_HLW_CF = 22;
   int GPIO_NRG_CF1 = 24;

   bool g_sel = true;
   ...
   ~~~
- After that I compiled the App and I got success, generating the file `OpenBK7231N_App_1.0.0.rbl`.
- Then I flashed the new compiled firmware via OTA and it all went OK, restarting the device.
- Everything seems to work OK except now when I issue the custom command `startDriver BL0937`, the device either freezes (most of the times) or restarts (some times). I think there must be something else apart from the GPIOS definition.
- This is what I get on the logs before the device freezes:

   ~~~
   Debug:CMD:cmd [startDriver BL0937]
   Debug:CMD:Adding command PowerSet
   Debug:CMD:Adding command VoltageSet
   Debug:CMD:Adding command CurrentSet
   Debug:CMD:Adding command PREF
   Debug:CMD:Adding command VREF
   Debug:CMD:Adding command IREF
   Info:NTP:Started BL0937.
   ~~~

I don't know what may be the problem with this driver, but if you can shred some light onto it I can check whatever you want and I can compile the changes and test them.

Thanks in advance.

openshwprojects, 2022-06-04:
Ok so two things
1. BK7231N has a bug in SDK which causes device to freeze if you do printf("%f" or addLogPrintf("%f" - just using %f in print hangs device
Most likely this line breaks:
	addLogAdv(LOG_INFO, LOG_FEATURE_BL0942,"Voltage %f, current %f, power %f\n", final_v, final_c, final_p);
	comment it out.
2. I was going to add ability to choose BL pins from GUI so there is no need to hardcode stuff... you can test on your own, but I will add a solution for your problems really soon

ec-blaster, 2022-06-04:
Indeed!

I made this change:

~~~ c
...
        //addLogAdv(LOG_INFO, LOG_FEATURE_BL0942,"Voltage %f, current %f, power %f\n", final_v, final_c, final_p);
	char res[128];
	sprintf(res,"Voltage %f, current %f, power %f\n", final_v, final_c, final_p);
	addLogAdv(LOG_INFO, LOG_FEATURE_BL0942,res );
...
~~~

and now it doesn't freeze nor reset. Indeed it's reading values and publishing them via MQTT. Thanks!

This is my log now:

~~~
Info:BL0942:Voltage pulses 1678, current 2, power 1
Info:BL0942:Voltage 222.385544, current 0.023715, power 1.500000
Info:MAIN:Time 274, free 91216, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38
Info:BL0942:Voltage pulses 1678, current 2, power 1
Info:BL0942:Voltage 222.385544, current 0.023715, power 1.500000
Info:MAIN:Time 275, free 91216, MQTT 1, bWifi 1, secondsWithNoPing 2, socks 2/38
Info:BL0942:Voltage pulses 1681, current 2, power 0
Info:BL0942:Voltage 222.783142, current 0.023715, power 0.000000
Info:MQTT:Publishing power = 0.000000 
Info:MAIN:Time 276, free 90752, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38
Info:BL0942:Voltage pulses 1681, current 2, power 1
Info:BL0942:Voltage 222.783142, current 0.023715, power 1.500000
Info:MQTT:Publishing power = 1.500000 
Info:MAIN:Time 277, free 90976, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38
~~~

And these are the MQTT messages:

[image]

So, I'm going to calibrate now the device and integrate it in Home Assistant. And I keep waiting (impatiently) to the moment you have time to deal with the solution you have on mind. I'll write an article in elektroda as soon as I flash your fixed firmware.

Thanks a lot!

JanKiel2000, 2022-06-04:
It works.
Only "Power" is 0.00000 or 1.500000
It can now add to Domoticz. It can even be published in parallel from MQTT to OH.
As I wrote in another thread I have other GPIO and I also have to compile myself.

openshwprojects, 2022-06-04:
I will push tomorrow commit that allows you to set pins via web gui, please wait just that night.

@JanKiel2000 are you still usind Domoticz? I will try to add Domoticz support soon. What would you suggest for testing? A VMware with Domoticz would suffice? What kind of Domoticz version do you run, and is there something else I need to be aware of? I will try to add the MQTT format we discussed several weeks ago in another issue.

JanKiel2000, 2022-06-04:
That's how I use Domoticz. :)
Domoticza, I have the latest version that I complicate from the sources.
You can use a virtual machine with Linux and install Domoticz.

openshwprojects, 2022-06-05:
@JanKiel2000 @ec-blaster  can you both check this commit? https://github.com/openshwprojects/OpenBK7231T_App/commit/cff8c71b4f49543497eacaea3f169bbafd2a4da6 i am not able to test with bl0937 right now but I wanted to get it to you fast

JanKiel2000, 2022-06-05:
Hello.
However, there is a mistake.
New commands for initialization should be in the "Init" procedure and not "Runframe".
It works after moving.
This is how it should look

void BL0937_Init() {
<------>// if not found, this will return the already set value
<------>GPIO_HLW_SEL = PIN_FindPinIndexForRole(IOR_BL0937_SEL,GPIO_HLW_SEL);
<------>GPIO_HLW_CF = PIN_FindPinIndexForRole(IOR_BL0937_CF,GPIO_HLW_CF);
<------>GPIO_HLW_CF1 = PIN_FindPinIndexForRole(IOR_BL0937_CF1,GPIO_HLW_CF1);
<------>HAL_PIN_Setup_Output(GPIO_HLW_SEL);
    HAL_PIN_SetOutputValue(GPIO_HLW_SEL, g_sel);

<------>HAL_PIN_Setup_Input_Pullup(GPIO_HLW_CF1);

    gpio_int_enable(GPIO_HLW_CF1,IRQ_TRIGGER_FALLING_EDGE,HlwCf1Interrupt);

<------>HAL_PIN_Setup_Input_Pullup(GPIO_HLW_CF);
    gpio_int_enable(GPIO_HLW_CF,IRQ_TRIGGER_FALLING_EDGE,HlwCfInterrupt);

<------>CMD_RegisterCommand("PowerSet","",BL0937_PowerSet, "Sets current power value for calibration", NULL);
<------>CMD_RegisterCommand("VoltageSet","",BL0937_VoltageSet, "Sets current V value for calibration", NULL);
<------>CMD_RegisterCommand("CurrentSet","",BL0937_CurrentSet, "Sets current I value for calibration", NULL);
<------>CMD_RegisterCommand("PREF","",BL0937_PowerRef, "Sets the calibration multiplier", NULL);
<------>CMD_RegisterCommand("VREF","",BL0937_VoltageRef, "Sets the calibration multiplier", NULL);
<------>CMD_RegisterCommand("IREF","",BL0937_CurrentRef, "Sets the calibration multiplier", NULL);
}
void BL0937_RunFrame() {
<------>float final_v;
<------>float final_c;
<------>float final_p;
<------>if(g_sel) {
<------><------>res_v = g_vc_pulses;
<------><------>g_sel = false;
<------>} else {
<------><------>res_c = g_vc_pulses;
<------><------>g_sel = true;
<------>}
    HAL_PIN_SetOutputValue(GPIO_HLW_SEL, g_sel);
<------>g_vc_pulses = 0;

<------>res_p = g_p_pulses;
<------>g_p_pulses = 0;

<------>addLogAdv(LOG_INFO, LOG_FEATURE_BL0942,"Voltage pulses %i, current %i, power %i\n", res_v, res_c, res_p);

<------>final_v = res_v * calib_v;
<------>final_c = res_c * calib_c;
<------>final_p = res_p * calib_p;
<------>{
<------><------>char dbg[128];
<------><------>sprintf(dbg,"Voltage %f, current %f, power %f\n", final_v, final_c, final_p);
<------><------>addLogAdv(LOG_INFO, LOG_FEATURE_BL0942,dbg);
<------>}

<------>BL_ProcessUpdate(final_v,final_c,final_p);
}
After how I turn on/off the relay, it will freeze for a moment.

openshwprojects, 2022-06-05:
@JanKiel2000 I wouldn't say that is a mistake. It was intentionally so users, during testing, can change configuration at runtime.

Are you saying that having it in runframe breaks things? I can move it to init, if that's needed

JanKiel2000, 2022-06-05:
When the initialization commands are in the "runframe" it shows all zeros for me.  Values ​​appear only after transferring to "init".  You just have to call the "init" routine after each change to the pins.

openshwprojects, 2022-06-05:
Ok i know why, simply the init call for gpio was missing. Fixed, for now.

JanKiel2000, 2022-06-05:
could add automatic activation or deactivation of the driver.  For example, when selecting GPIO in GUI, the condition of selecting three GPIOs for BL0937 will be met, the command to activate the driver will be automatically sent, otherwise the driver will be deactivated.

openshwprojects, 2022-06-05:
Great sugestion, I will add this tomorrow

ec-blaster, 2022-06-05:
Ok, I've been out for the weekend and I just read about your improvements.
I can confirm, as @JanKiel2000, that it works fine now. I still have to do some more testing, as for now I have a simple LED bulb connected and the power readings vary all the time between 0W and 1.5W... I suppose that with higher loads I could calibrate it and have more stable results.

As for the improvements, it's a good idea to have the driver auto activate on demand.

Thanks for your work.

JanKiel2000, 2022-06-05:
As for the consumed power. So "CF" I think that this value should not be deleted. It should be remembered and possible to zero a suitable command. It turns out that the correct pattern is
"Final_P = Final_P + (Res_P * Calib_P / 3600000);"
Where "Final_P" we define a driver at the beginning.
volatile uint32_t g_vc_pulses = 0;
volatile uint32_t g_p_pulses = 0;
float final_p = 0; // Read from Store
And we remove from the procedure.
I found something like this here
https://www.belling.com.cn/media/file_object/bel_product/bl0937/guide/bl0937%20App%20note_V1.0_en_en.pdf

JanKiel2000, 2022-06-06:
This could be used to make daily and total electricity consumption.

JanKiel2000, 2022-06-06:
> I am not sure what exactly are you trying to do. The final P (and "power") is currently defined as a momentary power consumption (for example, if we connect 60W bulb, it should say about 60W). The change you propose is just summing up the power consumption every frame. If I do your change, I will never know how much W takes 60W bulb at the moment.

OK. There is no topic. You're right.

ec-blaster, 2022-06-06:
I think we could close the issue, don't we?

closed 2022-06-07 by ec-blaster