openshwprojects/OpenBK7231T_App #93: PowerOnState is supported?

brijins, 2022-06-07:
Hi, 

I would like to know if PowerOnState (tasmota command) feature is supported, if it is, how to change its value? what I have is a smart plug and I would like it to turn on the relay as soon as it boots up. 

another minor issue I noticed is the state of the switch is not updated in HA once the device is powered back on, after a powerloss. if that can be fixed somehow, it would be nice. 

thank you!

openshwprojects, 2022-06-07:
Good questions! I have updated readme:
https://github.com/openshwprojects/OpenBK7231T_App/commit/f4fe2d8fad5481370906fb032267034bec136da2
Is this what you need?

The "save power state" will come soon, it's not yet done.

Hmmmm switch is not updated... maybe you are right, I will try to fix it today, will you check if it works later? Thanks|!

openshwprojects, 2022-06-07:
It seems that you basically need me to broadcast all channel values on MQTT init.

Ok, give me a second

openshwprojects, 2022-06-07:
I must investigate it.

brijins, 2022-06-08:
Hi,

1.4.9, no changes observed in behavior. 

I can share logs from the device if that is helpful in someway..

`fo:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING

Info:MAIN:Time 9, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 10, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:GEN:dhcp=0 ip=0.0.0.0 gate=0.0.0.0 mask=0.0.0.0 mac=d8:1f:12:51:14:b5 

Info:MQTT:mqtt_userName admin

mqtt_pass ********

mqtt_clientID heaterw16a

mqtt_host 192.168.50.59:1883

Info:MQTT:Connect error in mqtt_client_connect - code: -4

Info:MAIN:Time 11, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 12, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 13, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 14, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 15, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 16, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 17, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 18, free 92104, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:wl_status 9

Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_AUTH_FAILED

Info:MAIN:Time 19, free 91408, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 20, free 91408, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:GEN:dhcp=0 ip=0.0.0.0 gate=0.0.0.0 mask=0.0.0.0 mac=d8:1f:12:51:14:b5 

Info:MAIN:Time 21, free 90912, MQTT 0, bWifi 0, secondsWithNoPing 0, socks 2/38

Info:MAIN:wl_status 3

Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING

Info:MAIN:wl_status 10

Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED

Info:MAIN:wl_status 11

Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED

Info:MQTT:mqtt_userName admin

mqtt_pass ********

mqtt_clientID heaterw16a

mqtt_host 192.168.50.59:1883

Info:MAIN:Time 22, free 91368, MQTT 0, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MQTT:mqtt_connection_cb: Successfully connected

Info:MQTT:mqtt_subscribed to heaterw16a/+/set

Info:MQTT:mqtt_subscribed to cmnd/heaterw16a/+

Info:MQTT:MQTT client "heaterw16a" request cb: err 0

Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic heaterw16a/1/set

Info:MQTT:MQTT in topic heaterw16a/1/set

Info:MQTT:channelSet topic 4204460

Info:MQTT:channelSet part topic 1/set

Info:MQTT:channelSet channel 1

Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 1

Info:GEN:CHANNEL_Set channel 1 has changed to 1 (flags 0)

Info:MAIN:Channel has changed! Publishing change 1 with 1 

Info:MQTT:Publishing 1 = 1 

Info:MQTT:MQTT client "heaterw16a" request cb: err 0

Info:MAIN:Forced channel publish! Publishing val 0 with 0 

Info:MQTT:Publishing 0 = 0 

Info:MAIN:Forced channel publish! Publishing val 1 with 1 

Info:MQTT:Publishing 1 = 1 

Info:MAIN:Time 23, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Forced channel publish! Publishing val 1 with 1 

Info:MQTT:Publishing 1 = 1 

Info:MAIN:Time 24, free 91408, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 25, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 26, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 27, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 28, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 29, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 30, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:GEN:dhcp=0 ip=192.168.50.34 gate=192.168.50.1 mask=255.255.255.0 mac=d8:1f:12:51:14:b5 

Info:MAIN:Time 31, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Boot complete time reached (30 seconds)

Info:CFG:####### Set Boot Complete #######

Debug:CFG:new offset 764, boot_count 50, success count 50

Debug:CFG:new offset after read 764, boot_count 50, success count 50

Debug:CFG:re-read - offset 764, boot count 50, boot success 50, bootfailures 0
Info:MAIN:Time 32, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 33, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 34, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 35, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 36, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 37, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 38, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 39, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38

Info:MAIN:Time 40, free 91616, MQTT 1, bWifi 1, secondsWithNoPing 0, socks 2/38`

openshwprojects, 2022-06-08:
I already pushed a new version so it's hard to say about the obsolete one.

https://www.elektroda.com/rtvforum/viewtopic.php?p=20052191#20052191
https://github.com/openshwprojects/OpenBK7231T_App/commit/511af61218d912d8aafc16c18a5d6ee2c26c18dc

brijins, 2022-06-08:
will try and update the status soon

brijins, 2022-06-08:
1.4.10 worked as expected!! The only difference is I have "Rel_n" configured as pin so I had to use "1" as startup value, it is working as expected! thank you! 🥇 HA is also getting updated, I can sleep peacefully now 👍

closed 2022-06-08 by brijins