openshwprojects/OpenBK7231T_App #94: BK7231N+BL0937 The device hangs.

JanKiel2000, 2022-06-07:
When it activates the controller from BL0937 after some time (about an hour) the device stops working. There is nothing special in the logs. I had a preview in logs through the www all the time.
Log tip:

Info:MAIN:Time 4986, free 91368, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:BL0942:Voltage pulses 1803, current 0, power 0
Info:BL0942:Voltage 238.951813, current 0.000000, power 0.000000
Info:MAIN:Time 4987, free 91368, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MQTT:Publish result: -3
Info:BL0942:Voltage pulses 1803, current 0, power 0
Info:BL0942:Voltage 238.951813, current 0.000000, power 0.000000
Info:MAIN:Time 4988, free 91368, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:BL0942:Voltage pulses 1803, current 0, power 0
Info:BL0942:Voltage 238.951813, current 0.000000, power 0.000000
Info:MAIN:Time 4989, free 91368, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:BL0942:Voltage pulses 1803, current 0, power 0
Info:BL0942:Voltage 238.951813, current 0.000000, power 0.000000
Info:MAIN:Time 4990, free 91368, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=10.0.0.46 gate=10.0.0.1 mask=255.255.255.0 mac=c8:47:8c:01:64:99 
Info:BL0942:Voltage pulses 1807, current 0, power 0
Info:BL0942:Voltage 239.481934, current 0.000000, power 0.000000
Info:MAIN:Time 4991, free 91048, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:BL0942:Voltage pulses 1807, current 0, power 0

Info:BL0942:Voltage 239.481934, current 0.000000, power 0.000000

Info:MAIN:Time 4992, free 91368, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38

Info:MQTT:Publish result: -3

openshwprojects, 2022-06-07:
Was it always happening or has it started after latest updates?

JanKiel2000, 2022-06-07:
Now it crashed again.  On the previous "FV" I did not check because there was no support blah BL0937.  Now it worked 4 hours, I restarted, activated the BL0937 driver and after 3 hours the device freezes.  There is no contact with it, even the button does not work.  FV 1.4.4

openshwprojects, 2022-06-08:
i will leave it running today and see
[image]

JanKiel2000, 2022-06-08:
After a night, the device with BL0937 active also crashed.  Now it tests without activating this driver.

openshwprojects, 2022-06-08:
if device does not show up on your wifi, it might have booted into safe mode and act as open access point. Check this.

still everying ok on my side:
[image]

JanKiel2000, 2022-06-09:
Maybe it's just a problem with BL0937?  Restore the device and check it with the driver activated and without activating it.

JanKiel2000, 2022-06-09:
And it broke.
I had to restart.
[image]

JanKiel2000, 2022-06-09:
The device does not start, but hangs.  Now without a driver until 0937 it works for more than 2 hours.  Maybe a problem with the framework for "N"?

ec-blaster, 2022-06-10:
I must say that I've been reading this issue and I was to say that mine was stable (equivalent configuration, T34 and BL0937). But finally, after a window between 24h and 48h, the device hanged. It was on firmware 1.4.9.
I'll flash the latest version and check again.
PS: I landed at this repo because my device, with the original Tuya firmware, had these same symptons: it hanged between 24 and 48 hours. I thought this firmware may solve it. I have the ping watchdog configured for 30 seconds.

openshwprojects, 2022-06-10:
@ec-blaster  original Tuya firmware hangs as well?

ec-blaster, 2022-06-10:
Yes it does

openshwprojects, 2022-06-11:
Then it's most likely the hardware issue. Maybe bad capacitor?

I recently wrote a guide about that, but English translation is not done yet. Use google translate:
https://www.elektroda.pl/rtvforum/viewtopic.php?t=3898805&highlight=

JanKiel2000, 2022-06-11:
For me, the device works quietly.  So far the only hang up is when using the measurement module.  Also not right away.

openshwprojects, 2022-06-11:
@JanKiel2000  was it also crashing on original tuya firmware?

JanKiel2000, 2022-06-11:
I do not know.  I changed to OpenBK right away :)

JanKiel2000, 2022-06-12:
I guess that's the problem.  After a while, I have such lags.  And then the device freezes.

Info:MQTT:Publishing voltage = 234.710846 

Info:MQTT:Publish err: ERR_MEM aka -1

openshwprojects, 2022-06-12:
@JanKiel2000 I know about this error, it happens when you have too many MQTT publishes at the same time. I tried to increase the limits but it still happens in some rare cases.

Still, I have never experienced it crashing

I am no sure if it 's the cause, was it always in your log just before crash?

JanKiel2000, 2022-06-12:
I will still be testing.

ec-blaster, 2022-06-12:
I'll test it too.

My results with v1.4.16: 23 hours alive and then again hanged. I'm going to disable the power meter and test again with the same version, let's see if it's stable that way.

ec-blaster, 2022-06-12:
> Then it's most likely the hardware issue. Maybe bad capacitor?
> 
> I recently wrote a guide about that, but English translation is not done yet. Use google translate:
> https://www.elektroda.pl/rtvforum/viewtopic.php?t=3898805&highlight=

The device is new, and it makes no perceptible noise. I'll check the capacitors anyway.

openshwprojects, 2022-06-12:
I am getting N chip BL0937 socket and will check this ou tas well.

@ec-blaster  you also have N version?

ec-blaster, 2022-06-12:
> I am getting N chip BL0937 socket and will check this ou tas well.
> 
> @ec-blaster  you also have N version?

I have a T34, a superset of N

JanKiel2000, 2022-06-12:
One more lead.  I noticed that it also hangs when I turn on and off the relay a few times via the GUI.  The address "http://10.0.0.46/index?tgl=0".  Of course, not immediately, but after an hour, for example.  If I do not use a relay, the device can work all night long.

openshwprojects, 2022-06-13:
agani with Info:MQTT:Publish err: ERR_MEM aka -1?

JanKiel2000, 2022-06-13:
This error comes and goes.  However, it may not be the cause of the problem.

openshwprojects, 2022-06-13:
@JanKiel2000  are you connected to MQTT server? I remember you are a domoticz user.

does error happen if you set MQTT credentials to none, so MQTT is not running/.?

JanKiel2000, 2022-06-13:
MQTT was set but it went in a vacuum ... Nothing responded to the publications. I removed IP now with mqtt.

JanKiel2000, 2022-06-21:
I think I already know what's going on.  As I have everything set to channel 0 in the settings, it freezes then.  As soon as I change to channel 1, i.e. I will give 1 in instead of 0, I already have over 5 hours of device operation.

openshwprojects, 2022-06-21:
@JanKiel2000  can you explain in more detail? Channel 0 is not special in any way. The used channel index should not affect (at least in theory) the device operation. There might be some bug that I need to find.

JanKiel2000, 2022-06-21:
False alarm.  At 12 no more contact with the device.  He worked all night and then he collapsed.

ec-blaster, 2022-06-24:
> I'm still testing stability...
> 
> [image]
> 
> ... and v1.4.27 seems to perform pretty well.
> 
> 

It hanged after 4 days

JanKiel2000, 2022-06-25:
I have restricted the reading of sensor data.  I now have a reading every 10 seconds.  The device has been in operation for 14 hours.  We'll see what's next.

JanKiel2000, 2022-07-04:
I probably found a problem with hanging the device. The device works over 1 day without suspension. Login in "Bl0937_Runframe" must be turned off.
It is supposed to look like this.

void BL0937_RunFrame() {
<------>float final_v;
<------>float final_c;
<------>float final_p;

<------>if(g_sel) {
<------><------>res_v = g_vc_pulses;
<------><------>g_sel = false;
<------>} else {
<------><------>res_c = g_vc_pulses;
<------><------>g_sel = true;
<------>}
    HAL_PIN_SetOutputValue(GPIO_HLW_SEL, g_sel);
<------>g_vc_pulses = 0;

<------>res_p = g_p_pulses;
<------>g_p_pulses = 0;

//<---->addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"Voltage pulses %i, current %i, power %i\n", res_v, res_c, res_p);

<------>final_v = res_v * calib_v;
<------>final_c = res_c * calib_c;
<------>final_p = res_p * calib_p;
<------>{
//<----><------>char dbg[128];
//<----><------>sprintf(dbg,"Voltage %f, current %f, power %f\n", final_v, final_c, final_p);
//<----><------>addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,dbg);
<------>}

<------>BL_ProcessUpdate(final_v,final_c,final_p);
}

JanKiel2000, 2022-07-05:
I probably found a problem with hanging the device. The device works over 1 day without suspension. Login in "Bl0937_Runframe" must be turned off.
It is supposed to look like this.

void BL0937_RunFrame() {
<------>float final_v;
<------>float final_c;
<------>float final_p;

<------>if(g_sel) {
<------><------>res_v = g_vc_pulses;
<------><------>g_sel = false;
<------>} else {
<------><------>res_c = g_vc_pulses;
<------><------>g_sel = true;
<------>}
HAL_PIN_SetOutputValue(GPIO_HLW_SEL, g_sel);
<------>g_vc_pulses = 0;

<------>res_p = g_p_pulses;
<------>g_p_pulses = 0;

//<---->addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"Voltage pulses %i, current %i, power %i\n", res_v, res_c, res_p);

<------>final_v = res_v * calib_v;
<------>final_c = res_c * calib_c;
<------>final_p = res_p * calib_p;
<------>{
//<----><------>char dbg[128];
//<----><------>sprintf(dbg,"Voltage %f, current %f, power %f\n", final_v, final_c, final_p);
//<----><------>addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,dbg);
<------>}

<------>BL_ProcessUpdate(final_v,final_c,final_p);
}

PS.
How is it with support for Domoticz. How can I test it.

openshwprojects, 2022-07-09:
Hey, any news on this? Is it stable now?

JanKiel2000, 2022-07-09:
It sure is better.  Tests two devices in one I only deleted this command.  I uploaded the second one with other corrections.  This was the first time that it crashed.  I check in a longer period.
1.
[image]
2.
[image]

JanKiel2000, 2022-07-09:
[image]

JanKiel2000, 2022-07-09:
[image]

JanKiel2000, 2022-07-09:
[image]

JanKiel2000, 2022-07-09:
Approx. Jedank is a problem when I only remove these logging lines. The device hangs after 1 day and 12 hours. I still have this controller improved by me. Here, Taj has a device to operate over 2 days and works so far. Corrected controller in the attachment. I also tried to add a stopping function there but it happened.

`#include "../new_common.h"
#include "../new_pins.h"
#include "../new_cfg.h"
// Commands register, execution API and cmd tokenizer
#include "../cmnds/cmd_public.h"
#include "../mqtt/new_mqtt.h"
#include "../logging/logging.h"
#include "../hal/hal_pins.h"
#include "drv_local.h"
#include "drv_uart.h"
#include "../httpserver/new_http.h"

#include "BkDriverTimer.h"
#include "BkDriverGpio.h"
#include "sys_timer.h"
#include "gw_intf.h"

// HLW8012 aka BL0937
#define E_001E_GAIN 4
#define ELE_HW_TIME 1
#define HW_TIMER_ID 0

// Those can be set by Web page pins configurator
// The below are default values for Mycket smart socket
int GPIO_HLW_SEL = 24; // pwm4
int GPIO_HLW_CF = 7;
int GPIO_HLW_CF1 = 8;

bool g_sel = true;
uint32_t res_v = 0;
uint32_t res_c = 0;
uint32_t res_p = 0;
double calib_v = 0.13253012048f;
double calib_p = 1.5f;
double calib_c = 0.0118577075f;
bool firstinitial = true;
volatile uint32_t g_vc_pulses = 0;
volatile uint32_t g_p_pulses = 0;
int TimeOld = 0;

void HlwCfInterruptNew(unsigned char pinNum) {  // Service Power
	int curPin = (int)pinNum;
	if (curPin == GPIO_HLW_CF1) g_vc_pulses++;
	if (curPin == GPIO_HLW_CF) g_p_pulses++;
}

int BL0937_PowerSet(const void *context, const char *cmd, const char *args, int cmdFlags) {
	float realPower;

	if(args==0||*args==0) {
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"This command needs one argument");
		return 1;
	}
	realPower = atof(args);
	calib_p = realPower / res_p;
	{
		char dbg[128];
		sprintf(dbg,"CurrentSet: you gave %f, set ref to %f\n", realPower, calib_p);
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,dbg);
	}
	return 0;
}
int BL0937_PowerRef(const void *context, const char *cmd, const char *args, int cmdFlags) {

	if(args==0||*args==0) {
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"This command needs one argument");
		return 1;
	}
	calib_p = atof(args);
	return 0;
}
int BL0937_CurrentRef(const void *context, const char *cmd, const char *args, int cmdFlags) {

	if(args==0||*args==0) {
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"This command needs one argument");
		return 1;
	}
	calib_c = atof(args);
	return 0;
}
int BL0937_VoltageRef(const void *context, const char *cmd, const char *args, int cmdFlags) {

	if(args==0||*args==0) {
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"This command needs one argument");
		return 1;
	}
	calib_v = atof(args);
	return 0;
}
int BL0937_VoltageSet(const void *context, const char *cmd, const char *args, int cmdFlags) {
	float realV;

	if(args==0||*args==0) {
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"This command needs one argument");
		return 1;
	}
	realV = atof(args);
	calib_v = realV / res_v;
	{
		char dbg[128];
		sprintf(dbg,"CurrentSet: you gave %f, set ref to %f\n", realV, calib_v);
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,dbg);
	}

	return 0;
}
int BL0937_CurrentSet(const void *context, const char *cmd, const char *args, int cmdFlags) {
	float realI;

	if(args==0||*args==0) {
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,"This command needs one argument");
		return 1;
	}
	realI = atof(args);
	calib_c = realI / res_c;
	{
		char dbg[128];
		sprintf(dbg,"CurrentSet: you gave %f, set ref to %f\n", realI, calib_c);
		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,dbg);
	}
	return 0;
}
void BL0937_Init() {

	// if not found, this will return the already set value
	GPIO_HLW_SEL = PIN_FindPinIndexForRole(IOR_BL0937_SEL,GPIO_HLW_SEL);
	GPIO_HLW_CF = PIN_FindPinIndexForRole(IOR_BL0937_CF,GPIO_HLW_CF);
	GPIO_HLW_CF1 = PIN_FindPinIndexForRole(IOR_BL0937_CF1,GPIO_HLW_CF1);

	HAL_PIN_Setup_Output(GPIO_HLW_SEL);
	HAL_PIN_SetOutputValue(GPIO_HLW_SEL, g_sel);

//	HAL_PIN_Setup_Input_Pullup(GPIO_HLW_CF1);
	gpio_int_enable(GPIO_HLW_CF1,IRQ_TRIGGER_FALLING_EDGE,HlwCfInterruptNew);
//	HAL_PIN_Setup_Input_Pullup(GPIO_HLW_CF);
	gpio_int_enable(GPIO_HLW_CF,IRQ_TRIGGER_FALLING_EDGE,HlwCfInterruptNew);
	TimeOld = Time_getUpTimeSeconds();
	if (firstinitial)
	{
		CMD_RegisterCommand("PowerSet","",BL0937_PowerSet, "Sets current power value for calibration", NULL);
		CMD_RegisterCommand("VoltageSet","",BL0937_VoltageSet, "Sets current V value for calibration", NULL);
		CMD_RegisterCommand("CurrentSet","",BL0937_CurrentSet, "Sets current I value for calibration", NULL);
		CMD_RegisterCommand("PREF","",BL0937_PowerRef, "Sets the calibration multiplier", NULL);
		CMD_RegisterCommand("VREF","",BL0937_VoltageRef, "Sets the calibration multiplier", NULL);
		CMD_RegisterCommand("IREF","",BL0937_CurrentRef, "Sets the calibration multiplier", NULL);
		firstinitial = false;
	}
}

void BL0937_Stop() {
	gpio_int_disable(GPIO_HLW_CF1);
	gpio_int_disable(GPIO_HLW_CF);
}

void BL0937_RunFrame() {
	int TimeNow = Time_getUpTimeSeconds();
	int delta = TimeNow - TimeOld;
	TimeOld = TimeNow;
	float final_v;
	float final_c;
	float final_p;
	if(g_sel) {
		res_v = g_vc_pulses;
	} else {
		res_c = g_vc_pulses;
	}
	g_sel = !g_sel;
	HAL_PIN_SetOutputValue(GPIO_HLW_SEL, g_sel);
	g_vc_pulses = 0;
	res_p = g_p_pulses;
	g_p_pulses = 0;

//		sprintf(dbg,"Voltage pulses %i, current %i, power %i\n", res_v, res_c, res_p);
//		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,dbg);
	final_v = res_v * calib_v / delta;
	final_c = res_c * calib_c / delta;
	final_p = res_p * calib_p / delta;
//		sprintf(dbg,"Voltage %f, current %f, power %f\n", final_v, final_c, final_p);
//		addLogAdv(LOG_INFO, LOG_FEATURE_BL09XX,dbg);
	BL_ProcessUpdate(final_v,final_c,final_p);
}

`

openshwprojects, 2022-07-10:
Hey, can you debug print the "delta" value in your code?

RunFrame is called every frame, which is every second, so "TimeNow-TimeOld" will always give you 1

I think it's not possible for "TimeNow-TimeOld" to be non-1 because the execution flow ALWAYS increases timer by 1 and then executes RunFrame for each driver

And then you will always do "res_v * calib_v / 1"
and it will always come down to res_v * calib_v

openshwprojects, 2022-07-10:
Also: what is the purpose of firstinitial  ?

DRV_StartDriver makes sure you dont start a driver twice....
[image]

JanKiel2000, 2022-07-10:
It was more that when you called the "stopDriver" function, it did not reinitialize the measurement commands on the next activation.  In my opinion, the most important changes are that there is one function that counts pulses, I switch interrupts a bit differently.

openshwprojects, 2022-07-10:
Also what is the purpose of "int curPin = (int)pinNum;"? It does nothing, comparison will work ok for two different types as well

JanKiel2000, 2022-07-10:
It was more that when you called the "stopDriver" function, it did not reinitialize the measurement commands on the next activation.  In my opinion, the most important changes are that there is one function that counts pulses, I switch interrupts a bit differently.
"(int) pin Num" converts "char" to "integer".  For comparing CF1 and CF.  I know the function is supposed to be every second, but if there was a delay.  You could read every 2 seconds, not one.

openshwprojects, 2022-07-10:
That's very interesting, I will investigate it, altough I have two more question:

_"It was more that when you called the "stopDriver" function, it did not reinitialize the measurement commands on the next activation_." you are indeed correct here, but do anyone expect restarting the driver in a normal use case?

_"I know the functions supposed to be every second, but if there was a delay. You could read every 2 seconds, not one."_ this, I think, is not correct, because the Time_getUpTimeSeconds is tied 1:1 to the RunFrame so it's not possible that delta value is different than 1
Or maybe you want to modify RunFrame to be called every 2 seconds, if so, then you are correct

**Ok, now please, check your version if tell me here if it crashes (tomorrow or in few days). I really don't expect that using single function with IF instead of two functions makes a difference, but let's see**

JanKiel2000, 2022-07-10:
As for the "delta", it remained after reading the data every 10sec. Then I was only able to achieve that the device was hanging later. This Stopdriver I was thinking about connecting to the changes of the controller and controller reset. Not everyone has to know the GPIO at their disposal immediately and can look for trial and mistakes.
The device works at the moment
"Online for 3 Days, 23 Hours, 27 Minutes and 15 Seconds"

openshwprojects, 2022-07-16:
@JanKiel2000  do you recognize this error message?
[image]
Is this what happens to your device?

l3ztum, 2022-07-18:
@openshwprojects i have the same hanging error with the same combination of components

currently on version 1.10.4
button is wired to rx of BK7231N
configuration: 
[image]
usually unresponsive after 4-5h [image]

openshwprojects, 2022-07-19:
@JanKiel2000  @l3ztum  I will need you both around in several days or maybe week for some testing.

The first thing to be tested will be the current update which is not released yet because I am testing it myself.

Then I will be updating LWIP to newer version, which also might help.

**Can you please tell me whether "dead" device replies to HTTP with ERR_EMPTY_RESPONSE like on the screenshot in my last post here?**

openshwprojects, 2022-07-24:
@l3ztum  what about current version?

Are you able to get the UART log from device, I mean, connect it to 5V (not MAINS !!), and keep the log to see what happens if it crashes?

l3ztum, 2022-07-25:
Updated yesterday to 1.11.2 but still got a freezing device while been plugged in. soldering rx on today to save log files to pc.

JanKiel2000, 2022-07-25:
I updated to the latest version I have my corrections and it hangs so on. Once after 3 days once after 12 hours. I am awaiting some tests and ideas.

l3ztum, 2022-07-26:
@openshwprojects
manged to test with rx2 and tx2 soldered to the CB2S Board. everything seemed to work fine until 6h and 30min in a error appeared.

> sm_disassoc_handler
> sm_disassoc_handler reason=2,vif=0
> sm_disconnect_process
> me_set_ps_disable:840 0 0 1 0 11
> SM_DISCONNECT_IND
> bad_channel vif_idx=0
> bad_channel vif_idx=0

l3ztum, 2022-08-13:
@openshwprojects problem persists on 1.12.21
testing 1.12.27 atm

l3ztum, 2022-08-15:
@openshwprojects same issue still hangs

openshwprojects, 2022-08-18:
I just had reports from Elektroda users suggesting that's a N-only chip issue here:
https://www.elektroda.com/rtvforum/topic3874289-120.html
**It is also said to be CB2S.**

It always ends with sm_disassoc_handler and SM_DISCONNECT_IND bad_channel vif_idx=0 messages?

It happens even without MQTT?

openshwprojects, 2022-08-18:
@l3ztum  so device after that is not even responding to buttons?
Is the counter still going after that hang?
Info:MAIN:Time 23435, free 78688, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

openshwprojects, 2022-08-18:
@l3ztum  are you able to compile yourself?

W:\GIT\OpenBK7231N\platforms\bk7231n\bk7231n_os\beken378\app\config\sys_config.h

/*section 24 ----- less memery in rwnx*/
#define CFG_LESS_MEMERY_IN_RWNX                    1

W:\GIT\OpenBK7231T\platforms\bk7231t\bk7231t_os\beken378\app\config\sys_config.h
#define CFG_LESS_MEMERY_IN_RWNX                    0

the N sdk has this enabled and T not, maybe we try to disable it for N as well? Are you able to compile binary yourself?

l3ztum, 2022-08-18:
@openshwprojects will try tomorrow, I can compile myself. rbl would be appreciated though because rx1 is connected to vcc via resistor on my board. had to desorder the smd resistor to flash.

[11 more replies]

closed 2022-11-04 by openshwprojects