openshwprojects/OpenBK7231T_App #95: Docs?

LuisThe0ne, 2022-06-19:
Hello,

I noticed that this project hasn't really got a documentation, so I wanted to ask if Docs are needed (@openshwprojects)?
I would really like to create them. because I want to be part of the project. (I would suggest to use Mkdocs, like tasmota does I think)

Regard,
Luis

TommySharpNZ, 2022-07-03:
Yeah I can't even figure out where to get BKwriter 1.6 so docs for us amateurs would be great!!! I'll happily help by contributing to them once I learn a bit more...

openshwprojects, 2022-07-03:
@TommySharpNZ  well, to be honest, there was a link to OpenBK7231T repo in the readme under bkWriter mention, and in that repo (in the same level) there is a bkWriter.zip, soo.... but as said in the other issue, now I added a direct link to repo, and it most likely won't get easier than that now.

LuisThe0ne, 2022-07-03:
> Hi Luis, a good place to start would be to populate the Wiki here. I really hate the TAS docs build process, but we could always migrate the pages & words if required later. The wiki should be publicly editable - and then it also means that you could create placeholders and ask for help to populate them? br, Simon

I think to pobulate the Wiki would be a ggreat Idea, so @openshwprojects I would be nice if you could make the wiki editablöe for everyone. (https://docs.github.com/en/communities/documenting-your-project-with-wikis/changing-access-permissions-for-wikis)

openshwprojects, 2022-07-14:
Hey, what's the state of this?
Wiki is already editable, I made the change the few days ago or so.

LuisThe0ne, 2022-07-14:
Sorry, did not notice the change I am currently on holiday with my family, but when I am back at home an I got access to my pc I am going to look into creating or starting the creation of a wiki.

LuisThe0ne, 2022-08-17:
The Wiki is now populated with I think most of the Information needed. :)

jekader, 2023-02-21:
Nice work on the documentation! I propose also shortening README.md and moving the rest of its current content to the /docs directory. The wiki is nice, but having docs in the repo is a good thing, too.

If you're fine with that I can prepare a PR doing this.

LuisThe0ne, 2023-02-21:
Sounds like a great idea!