openshwprojects/OpenBK7231T_App #97: Help with configuring to build

AcuarioCat, 2022-06-20:
I am trying to build this app for BK7231N on my own machine but I'm not getting very far.
I follow the instructions but they aren't that clear in some parts.
I end up with loads of 'No such file or directory messages.
Here are the 'problems'
1. I download the sdk as a zip file. The file is OpenBK7231N-master.zip. In the zip there is OpenBK7231N-master which then has various folders/files.
2. I unzip the folders and files to c:\bk7231sdk as in the readme so I end up with c:\openbk7231sdk\apps, platforms etc..
3. I download the app zip file. The file is OpenBK7231T_App-main.zip. In the zip there is OpenBK7231T_App-main which then has various folders/files.
4.  I unzip the folders and files to c:\bk7231sdk\apps as in the readme so I end up with c:\openbk7231sdk\apps\OpenBK7231T_App-main with various folders/files below it. I see from the b.sh this should be OpenBK7231T_App so I rename it.
5. I open cygwin prompt and go to /cygdrive/c/bk7231sdk
6. I run ./b.sh and get loads of file not found and Error 2

What am I doing wrong?
I have file structure:
c:\bk7231sdk
            apps
                 OpenBK7231N_App
                        .github
                        include
                        output
                        sdk
                        src
                 tuya_demo_elp_1plug
                 ......  3 other tuya apps
          platform

What should I have or what is wrong?

Thanks.

openshwprojects, 2022-06-21:
OpenBK7231T_App or OpenBK7231N_App ?

AcuarioCat, 2022-06-21:
I'm trying to build OpenBK7231N_App

Timo-Beil, 2022-06-21:
In the _N version the "tuya_common" folder is absent in /platforms/bk7231n/.
I took  it over from the -T version and the first error is fixed.
But I can not find, what the /components folder has to contain.

APP_PATH=apps/OpenBK7231N_App
APP_NAME=OpenBK7231N_App
APP_VERSION=1.0.0
USER_CMD=
APP_BIN_NAME=OpenBK7231N_App
APP_VERSION=1.0.0
TARGET_PLATFORM=bk7231n
USER_CMD=
Start Compile
system:CYGWIN_NT-10.0-22000
find: ‘../../../components’: No such file or directory
not is ci build

openshwprojects, 2022-06-21:
I am aware that there is a warning about missing components and tuya_common directories, but they are now non-existing and it compiles fine for me with those warnings.

Still, I modified the N SDK to stop asking for those dirs.

https://github.com/openshwprojects/OpenBK7231N/commit/8d5240f08503067acede2dffd7e7c528598c3bf3

Can you try now?

@Timo-Beil  @AcuarioCat 

[image]

AcuarioCat, 2022-06-21:
Where, which directory, should I be running the build from? I have tried from /cygdrive/c/bk7231sdk/apps/OpenBK7231N-App and I get a huge list of missing files as follow:

$ ./b.sh
APP_PATH=apps/OpenBK7231N_App
APP_NAME=OpenBK7231N_App
APP_VERSION=1.0.0
USER_CMD=
./build_app.sh: line 72: cd: apps/OpenBK7231N_App: No such file or directory
APP_BIN_NAME=OpenBK7231N_App
APP_VERSION=1.0.0
TARGET_PLATFORM=bk7231n
USER_CMD=
Start Compile
system:CYGWIN_NT-10.0-22000
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
not is ci build
make: Entering directory '/cygdrive/c/bk7231sdk/apps/OpenBK7231N-App/platforms/bk7231n/bk7231n_os'
make: git: No such file or directory
make[1]: Entering directory '/cygdrive/c/bk7231sdk/apps/OpenBK7231N-App/platforms/bk7231n/bk7231n_os'
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/src’: No such file or directory
echo prerequirement0
prerequirement0
===========================================================
Build OpenBK7231N_App
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
===========================================================
echo prerequirement1
prerequirement1
mkdir -p Debug/obj
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
mkdir -p Debug/bin
echo prerequirement2
prerequirement2
build beken378/app/app_bk.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/app/ate_app.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/app/config/param_config.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/app/standalone-ap/sa_ap.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/app/standalone-station/sa_station.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/demo/ieee802_11_demo.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/driver/common/dd.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/driver/common/drv_model.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
build beken378/driver/dma/dma.o
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
[... 870 lines cut ...]
prerequirement0
find: ‘../../../sdk/include’: No such file or directory
===========================================================
find: ‘../tuya_os_adapter/include’: No such file or directory
Build OpenBK7231N_App
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
===========================================================
find: ‘../../../sdk/include’: No such file or directory
echo prerequirement1
prerequirement1
find: ‘../tuya_os_adapter/include’: No such file or directory
mkdir -p Debug/obj
find: ‘../../../apps/OpenBK7231N_App/include’: No such file or directory
mkdir -p Debug/bin
echo prerequirement2
prerequirement2
build beken378/app/app_bk.o
find: ‘../../../sdk/include’: No such file or directory
/bin/sh: ../toolchain/windows/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: No such file or directory
find: ‘../tuya_os_adapter/include’: No such file or directory
make[1]: *** [application.mk:1073: beken378/app/app_bk.o] Error 127
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/cygdrive/c/bk7231sdk/platforms/bk7231n/bk7231n_os'
make: *** [Makefile:7: mp] Error 2
make: Leaving directory '/cygdrive/c/bk7231sdk/platforms/bk7231n/bk7231n_os'

openshwprojects, 2022-06-21:
Sorry for non-english OS, but take a look, here is my console log from working N build:

> openshw@DESKTOP-PC /cygdrive/w/GIT/OpenBK7231N
> $ ./b.sh
> APP_PATH=apps/OpenBK7231N_App
> APP_NAME=OpenBK7231N_App
> APP_VERSION=1.0.0
> USER_CMD=
> Calling original build_app.sh script
> cd /cygdrive/w/GIT/OpenBK7231N/apps/OpenBK7231N_App/../../platforms/bk7231n/toolchain/../bk7231n_os && sh build.sh OpenBK7231N_App 1.0.0 bk7231n
> APP_BIN_NAME=OpenBK7231N_App
> APP_VERSION=1.0.0
> TARGET_PLATFORM=bk7231n
> USER_CMD=
> Start Compile
> system:CYGWIN_NT-10.0
> not is ci build
> make[1]: Wejście do katalogu '/cygdrive/w/GIT/OpenBK7231N/platforms/bk7231n/bk7231n_os'
> make[1]: git: No such file or directory
> make[2]: Wejście do katalogu '/cygdrive/w/GIT/OpenBK7231N/platforms/bk7231n/bk7231n_os'
> echo prerequirement0
> prerequirement0
> ===========================================================
> Build OpenBK7231N_App
> ===========================================================
> echo prerequirement1
> prerequirement1
> mkdir -p Debug/obj
> mkdir -p Debug/bin
> build beken378/app/app_bk.o
> echo prerequirement2
> prerequirement2
> build beken378/app/ate_app.o
> build beken378/app/config/param_config.o
> build beken378/app/standalone-ap/sa_ap.o
> build beken378/app/standalone-station/sa_station.o
> build beken378/demo/ieee802_11_demo.o
> build beken378/driver/common/dd.o
> build beken378/driver/common/drv_model.o
> build beken378/driver/dma/dma.o
> build beken378/driver/driver.o
> build beken378/driver/entry/arch_main.o
> build beken378/driver/flash/flash.o
> build beken378/driver/general_dma/general_dma.o
> build beken378/driver/gpio/gpio.o
> build beken378/driver/i2c/i2c1.o
> build beken378/driver/i2c/i2c2.o
> build beken378/driver/icu/icu.o
> build beken378/driver/intc/intc.o
> build beken378/driver/irda/irda.o
> build beken378/driver/macphy_bypass/mac_phy_bypass.o
> build beken378/driver/phy/phy_trident.o
> build beken378/driver/pwm/pwm.o
> build beken378/driver/pwm/pwm_new.o
> beken378/driver/pwm/pwm_new.c: In function 'pwm_update_param':
> beken378/driver/pwm/pwm_new.c:820:1: warning: control reaches end of non-void function [-Wreturn-type]
>  }
>  ^
> beken378/driver/pwm/pwm_new.c: In function 'pwm_group_update_param_enable':
> beken378/driver/pwm/pwm_new.c:938:1: warning: control reaches end of non-void function [-Wreturn-type]
>  }
>  ^
> beken378/driver/pwm/pwm_new.c: In function 'pwm_nogroup_update_param_enable':
> beken378/driver/pwm/pwm_new.c:964:1: warning: control reaches end of non-void function [-Wreturn-type]
>  }
>  ^
> beken378/driver/pwm/pwm_new.c: In function 'pwm_param_clear':
> beken378/driver/pwm/pwm_new.c:1000:1: warning: control reaches end of non-void function [-Wreturn-type]
>  }
>  ^
[... 481 lines cut ...]
> image size: 1220464 Byte
> image name: all_1.00.bin
> 
> export app partition image: OpenBK7231N_App_1.0.0_enc_uart_1.00.bin
> 
> time: 4.82s
> Good bye!
> End Combined
> generate ota file
> sw_version:1.0.0
> OpenBK7231N_App_1.0.0.rbl file_size:0x79a50
> OpenBK7231N_App_1.0.0.rbl sum:0x3c9dc21
> header_sum:0x5a5
> rm OpenBK7231N_App_1.0.0.rbl
> ug_file size:
> 500208
> /cygdrive/w/GIT/OpenBK7231N/platforms/bk7231n/bk7231n_os/tools/generate
> *************************************************************************
> *************************************************************************
> *************************************************************************
> *********************OpenBK7231N_App_1.0.0.bin********************
> *************************************************************************
> **********************COMPILE SUCCESS************************************
> *************************************************************************
> not is ci build

openshwprojects, 2022-06-21:
I asked about that 6 hours ago, but I will ask again, because I am still suspicious.

Is your App directory called "OpenBK7231N_App" when compiling with N SDK? Or did you put a directory named "OpenBK7231T_App" in N sdk?

AcuarioCat, 2022-06-22:
I renamed the App to N but the code is from the T app.
Here is my directory structure. I see you have a bk7231n folder in the apps directory - what does this contain? And what do you have in the OpenBK7231N folder?
I'm sure it's just a case of the files/folders being in the wrong places.
Maybe if you could follow the instructions from the readme or what I did as described in my first post (which was what I understood from the readme) then you would see what is wrong. Possibly it is because I am downloading the zip files and not using the git desktop/ide/whatever?
[image]
[image]

openshwprojects, 2022-06-22:
I see what you did wrong and readme is correct.

Why do you have the same repo copied twice?

[image]

the SDK Repo for N: https://github.com/openshwprojects/OpenBK7231N
The App repo: https://github.com/openshwprojects/OpenBK7231T_App
The SDK repo for T: https://github.com/openshwprojects/OpenBK7231T

Timo-Beil, 2022-06-22:
Thanks, now it runs like a charme. 
My mistake. I hadn't put the "make" package in my Cygwin.

AcuarioCat, 2022-06-22:
I thought that was what I was doing, simply copying the contents from the zip to the correct directory.
Here is the contents of the SDK that I copy (extract) to c:\bk7231sdk
[image]
[image]

Next the contents of the app that I copy (extract) to the apps folder and rename from T to N

[image]
[image]

The app contains an SDK folder, I don't add anything, just extract from the zip

I see the app has folders that are in the sdk - apps, platform, sdk - I didn't create or add them.
I then try running the build (there are 2 b.sh scripts, one in the bk7231sdk folder and one in the bk7231sdk/apps/OpenBK7231_App folder) but neither work as noted above.

AcuarioCat, 2022-06-22:
ok, done - somehow yes it looks like they were mixed. Now it compiles. Next challenge understand it and modify for my system but that's another story.. thanks for your help.

closed 2022-06-22 by AcuarioCat

exzombie, 2022-11-06:
Posting here in case anyone has problems building and finds this closed issue: the errors shown in initial posts (like `/bin/sh: ../toolchain/.../bin/arm-none-eabi-gcc: No such file or directory`) indicate that the system doesn't have the right dynamic loader. For example, you are on 64-bit linux (which you surely are) and don't have a 32-bit libc installed. You need to install `libc6-i386` on Debian/Ubuntu to make it work. Still, I had issues running `otafix` on Ubuntu 22.04 which was crashing. And then there's the missing python problem …

The easiest solution is to build a docker container with an older Ubuntu. The following Dockerfile allowed me to build the images:
```
FROM ubuntu:20.04
    MAINTAINER Johann Hirschkuh <me@mymail.com>

env    DEBIAN_FRONTEND noninteractive
run    apt-get -q -y update && apt-get -q -y upgrade
run    apt-get -q -y install npm
run    apt-get -q -y install make
run    apt-get -q -y install libc6-i386
run    apt-get -q -y install python2 && ln -s /usr/bin/python2 /usr/local/bin/python
```

NorthernMan54, 2023-02-19:
@exzombie Tks for sharing this. With my builds I keep all the checked out code in my ~/Code directory.  So I did this

```
# Copy the repo's

cd ~/Code/
git clone https://github.com/openshwprojects/OpenBK7231T.git
cd OpenBK7231T/apps
git clone https://github.com/openshwprojects/OpenBK7231T_App.git

# Create the docker environment
mkdir ~/Docker/OpenBK7231T
cd ~/Docker/OpenBK7231T
# Copied the `Dockerfile` from above

# build the docker environment
docker build -t openbk .
docker run -it --mount src=~/Code,target=/Code,type=bind openbk bash

# Then inside the docker container
cd /Code/OpenBK7231T/
./b.sh

# After about 10 minutes, the binaries are here
cd /Code/OpenBK7231T/apps/OpenBK7231T_App/output/1.0.0
```