openshwprojects/OpenBK7231T_App #98: Battery-powered device support

pavel211, 2022-06-23:
Can your program work with battery-powered devices, such as a door opening sensor?

For example, this:  https://templates.blakadder.com/TY01.html

I bought several such sensors with CB3S controllers, but it is impossible to link them locally with the standard firmware, no TUYA-MQTT converters work, the device simply does not have time to respond to the request and turns off.

dskw, 2022-11-14:
bump because Tuya integration in HA is really crappy right now and the device is flashable, but lacks software support for deepsleep etc.

btsimonh, 2022-11-15:
for door sensors, I use the MI BLE version, 2032 lasts about a year.
I use TAS on ESP32 to listen to the BLE adverts....

bernd-wechner, 2023-01-17:
I have a set of Tuya sensors with CBU chips:

[image]

and I've logged their uptime reports from (using the Tuya developers portal):

[image]

That's with the standard firmware of course, and if reflashed we'd want comparable performance for battery conservation.

dskw, 2023-05-18:
I've managed to flash OpenBeken on my device. Turned out to be a door sensor withOUT tuyaMCU.
```
{
  "vendor": "Smart Life",
  "bDetailed": "0",
  "name": "Door/Window Sensor",
  "model": "n/a",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "8": "DoorSnsrWSleep_nPup;1",
    "24": "Btn;2",
    "26": "WifiLED_n;3"
  },
  "command": "",
  "image": "https://user-images.githubusercontent.com/5196016/194883132-312f95c5-e1df-4b15-8115-9b988219f9df.png",
  "wiki": ""
}
```

Using these startup commands:
```
addEventHandler OnHold 24 DSTime 2400
DSTime 5
DSEdge 0
```

So far it works as intended. The sensor updates the state in Homeassistant correctly.
Sensor works with deepsleep. LED is showing Wifi status and the button works for extending deepsleep for debugging use.

**Some issues remain though:**

Update time is ~18 seconds from sensor change to entering deep sleep.
The original firmware managed to do an update in ~1 second.

I was unable to find a way to read the battery charge. Tuya is able to tell in percent how much charge is left in the batteries, but I cannot find any pin with usable data. ADC is always 0.00 and I don't see any specific IC that would provide the information via 1-wire or similar.
Except for one labeled MG2 L2AA, but I could not find any information about this one.
[image]