openshwprojects/OpenBK7231T_App #100: Building test HTTP server with N SDK

AcuarioCat, 2022-06-27:
I'm trying to build the http server (myHTTP2022-2010) project using the N SDK. I'm using VS 2019.
When I open the solution there are several files that are reported as missing.
When I try to build I get an error 

Error	C1083	Cannot open include file: 'lwip/sockets.h': No such file or directory	myHTTP2022	C:\bk7231sdk\apps\OpenBK7231N_App\src\httpserver\rest_interface.c	14	

I see lwip/sockets.h is in the c:\bk7231sdk\sdk\include directory so I add the sdk directory to the VC++ include directories in the project properties.

Now I get cannot open souce file arch/cc.h (from arch.h) and lwipopts.h (from sockets.h)
I can't find either of these files in the sdk.

How can I fix this to be able to build the server?

AcuarioCat, 2022-06-27:
I was trying to build the server for testing. I can build the BK7231 firmware without problem.

openshwprojects, 2022-06-27:
So it's simply not maintained due to the lack of time. Would you like to try fixing the build of windows server and submitting a pull request?

Otherwise, I could do this somewhen in this or next week

AcuarioCat, 2022-06-27:
If I had a working starting point then I would be happy to try but as there are files missing that appear to be fundamental, and they aren't in either the T or N sdk then I wouldn't know what they contained. I imagine you have or had a working build so I guess you have a better chance than me in solving it.

openshwprojects, 2022-06-27:
Nope, all files are in place, it's just that one have to know which files are for which platform. For example, lwip/sockets are only for MCUs, on windows you can use WInsock header and that's all. It's just a matter of #ifdef defines and correct includes.

Still, it's some work to do. You'd also have to decide whether you want to only run http server on windows, or maybe also with whole stripped app skeleton..

anogma, 2022-07-08:
Before finding this issue, I also spent some hours trying to make it work. If I have time, I'll try to fix it. 

What about trying to compile with MINGW instead of VS2019? I like VS Code and it seems to work nicely with MINGW.

closed 2022-10-04 by openshwprojects