openshwprojects/OpenBK7231T_App #99: Power monitoring supported?

TurkeyMan, 2022-06-27:
I have some DIN relays with power monitoring.
It doesn't look like power monitoring is supported yet. I have 3-4 (similar) different brand devices here I can flash, is there a way I can help test?
I need power monitoring in addition to relay control.

PeterPolacek, 2022-06-27:
For me is bk7231n from NAS-WR07W powerplug, thank you

openshwprojects, 2022-06-27:
I have this plug, it has BL0942 or BL0937 so it's supported I think.
[image]
[image]

PeterPolacek, 2022-06-28:
Thank you, working now .. but voltage value is not correct .. i have 240V measured with volt meter in grid but plug shows me 257.9 .. how can i correct it? Thank you