openshwprojects/OpenBK7231T_App #102: BK7231N+BL0937 MQTT power data interval

PeterPolacek, 2022-06-30:
Hi, can i somewhere in web interface set interval for sending MQTT voltage, power and current data? I hope for static specific time, for example every 1 seconds.
thank you

valeklubomir, 2022-10-22:
Interval is fixed by the measurement logic to one sample per second. Sending of values is then triggered when minumum change of measured value is detected. Resulting interval is between 1s - 60s. Because there is how many samples can be skipped when changes are below threshold.

mariusbach, 2022-10-22:
Thanks! Is there a way to adjust the threshold?

valeklubomir, 2022-10-22:
Sample interval and Threshold are fixed in the firmware at moment.
We would need to understand the reason why to change threshold or interval? 
What benefit is being targetted?

mariusbach, 2022-10-22:
May I ask what the current threshold is? The idea would be to filter noise or unwanted power notifications at device level. 

For example monitoring a washing machine for being finished doesn't require many messages but just the big drop from full speed to done at the end. My machine then starts a keepalive cycle and turns every now and then. Setting a threshold higher than this cycling would not raise another "finished" message every time it turns again.

Implementing it in firmware always has the advantage, that everyone can use a reliable routine rather than having to program something on his own, with the chance of bugs etc.

jon-nfc, 2023-03-06:
+1 for this feature.

A use case for being able to set the reporting interval is to support calculating more accurate energy usage including usage costs. For example if the V/I/P is reported every second you can calculate power used relatively accurately with the following formula:

P_u = ((P / M) / S) 

then

P_u-count = P_u-count + P_u

Where:
P_u is total power used in Wh
P is measued power in Watts
M is minutes in one hour (60)
S is seconds in one minute (60)

P_u-count is the Wh used and starts with a value of '0' for the first calculation

Once you have P_u, you add it to the P_u-count to keep track of used Wh. P_u-count can now be used to calculate costs should you require it P_u-count x Cost per Kh.