openshwprojects/OpenBK7231T_App #103: MQTT Name with some topic

PeterPolacek, 2022-06-30:
Hi, when i  want to add some topic to mqtt message i can do it with adding topic/short_name .. but after reset device it stops sending mqtt messages at all.
Thank you

iprak, 2022-08-03:
The MQTT logging in the web Application might be useful in this regards.

I also opted to `Broadcast self state every minute` from under Flags. The setting mentions `May cause device disconnect's, DONT USE IT YET` but I haven't noticed any disconnect.
[image]

openshwprojects, 2023-02-22:
The MQTT disconnects were fixed thanks to great effforts of @valeklubomir  and @btsimonh . MQTT is now stable.

OpenBeken now support topics with slashes as well, like "beken/kitchen/lamp1", so it's also solved.

I am closing this issue now.

closed 2023-02-22 by openshwprojects