openshwprojects/OpenBK7231T_App #108: SC500W v3 - Curtain / Blind controller

Sineos, 2022-07-04:
Link: https://www.aliexpress.com/item/1005002710349987.html
Module: WB2S

Got this module and was pretty set back as I realized Tasmota was a no go. After some searching, found these great efforts here. Is there any chance to get this supported? Many thanks in advance.

[image]
[image]

openshwprojects, 2022-07-04:
It should be very easy, but the thing is, I don't have any such modules and I don't have any curtains with motors for testing.

Do you know C basics and would like to help out with the practical side of things?

Was it using ESP before?

Sineos, 2022-07-04:
Thanks for coming back to this.

> Do you know C basics and would like to help out with the practical side of things?

I'm not much of a developer. Bit of Node.js experience. Doing simple things works and of course, I'd be willing to support.

> Was it using ESP before?

Actually, no clue. There is https://www.aliexpress.com/item/1005001512832665.html, which I **think** is the predecessor and should be Tasmota able. Looks like they switched to their own development on v3 

> I could try following this guide: [tasmota.github.io/docs/Blinds-and-Shutters/#operation](https://tasmota.github.io/docs/Blinds-and-Shutters/#operation) and make it operate in the same way, but still, I wouldn't be able to test it myself

This would be great. From my understanding, key is that never ever both outputs are high at the same moment. This would kill the motor.

Sineos, 2022-07-12:
Flashed OpenBK and it is working so far:

[image]

From my point of view, a minimal Roller Shutter support would be:
* Make sure both relays are off at boot (already possible --> Configure Startup)
* Create an interlocking function: Before Relay 2 goes high, Relay 1 needs to be low and vice versa
* Create a "TimeToClose" / "TimeToOpen" setting for positioning. Personally I do not need any highly sophisticated calibration function. Anyway easier to time this manually. 
* Set / Return Position (0 - 100) via commands and MQTT 

Many thanks in advance for considering.

mkjustuk, 2022-10-13:
If I can help I'd like to, shutter/curtain support in OpenBeken would be a massive step.
I have this unit and a motor and switches to test the whole setup.  

This is a 2CH version, so has 4 relays and four switch outputs.

Link: [https://www.aliexpress.com/item/1005003661384708.html](https://www.aliexpress.com/item/1005003661384708.html)
Module: CBU - BK7231N - 7ori

I have several Tasmotized ESP based units already and I'd agree with Sineos on the basic requirements above, if you can mirror the Tasmota shutter support it's really great.

[image]
[image]
[image]