openshwprojects/OpenBK7231T_App #109: CSE7766 driver?

belikh, 2022-07-05:
Hi, is there any chance of including the power monitor IC CSE7766 driver?

I have this device: 
https://templates.blakadder.com/deta_6920HA.html

I've managed to flash the WB2S within and the touch sensors and relays are working, but I'm not sure how to proceed with adding voltage/current sensor.

It seems supported in Tasmota and ESPHome.

openshwprojects, 2022-07-05:
CSE7766 looks very simple, it uses UART interface, but I would need to get somehow one device for testing.

belikh, 2022-07-05:
Would remote web access to my device be of any help?    Otherwise I don't mind testing it.   I'm not sure how commonly the CSE7766 is used but we don't have a lot of options in Australia.

openshwprojects, 2022-07-06:
It would be very helpful, but I would have to have the remote access to openbk panel, I would need to be able to do a firmware update, and there would need to be some kind of load in the socket, maybe 10W bulb or something.

I would also need some photos from inside or at least know which pins are used by CSE7766

belikh, 2022-07-07:
I'm assuming that the chip is on the 1TX and 1RX pins, as I can't flash with those connected.   

Device is at http://110.151.54.191/ at least for now.  I'm connected via LTE temporarily.

belikh, 2022-07-07:
[image]

openshwprojects, 2022-07-07:
Ok let me see, but I think w should rather get a Discord or an email conversation

openshwprojects, 2022-07-07:
working on it

belikh, 2022-07-07:
Aha I have awoken to find Santa visited :p

I'll leave the port open until you say you don't need it :)

I've turned off the load though until you need it (energy crisis etc)

Thanks again!

openshwprojects, 2022-07-08:
@belikh  can you explain what exactly is inside this device, because I am not entirely sure? The power metering chip is measuring power consumption of the two sockets together?

belikh, 2022-07-08:
Correct.  It has one input,  two power outlets and two usb sockets.   I'm not sure if it's measuring usb also.  

It's a pretty tightly packed pcb and hard to tell what's going on.

I've got another I'll take some photos

openshwprojects, 2022-07-08:
@belikh  what is the voltage in your country, the exact measurement would be welcome...

Also, have you read that guide?
https://www.elektroda.com/rtvforum/topic3887748.html

belikh, 2022-07-08:
[image]

Does this help? :)

belikh, 2022-07-09:
The insides 

[image]
[image]
[image]
[image]

openshwprojects, 2022-07-09:
Thanks
I'll see if I get some free time today

openshwprojects, 2022-07-09:
Can you double-check some things?

1. read this guide
https://www.elektroda.com/rtvforum/topic3887748.html
2. here is your calibration:
[image]
3. do you also see that the Power goes to 0 with a certain delay after turning off the bulb? It's like it needs some time to notice there is no bulb
4. Do you also see that current consumption is still showing strange results even with both relays off? 
I am not sure about point 3, but point 4 can be fixed from software side by setting it to 0 when all relays are off...

belikh, 2022-07-09:
It seems like current is behaving opposite to expectation?

(relay state at top, blue is device reading, purple meross)

[image]

openshwprojects, 2022-07-10:
as the device is currently offline, can you try this and recalibrate?
https://github.com/openshwprojects/OpenBK7231T_App/commit/ce6f0aa37108b2c401c846ff876065c8acd98ff2

openshwprojects, 2022-10-16:
I think that @talltechdude tested this code and made some good adjustments, so now I can close this issue.

closed 2022-10-16 by openshwprojects