openshwprojects/OpenBK7231T_App #112: Please add some text to the introduction near the release information to explain what .rbl/QIO/UA/UG files are?

TurkeyMan, 2022-07-10:
Can you please add a little more text in the introduction where it mentions releases to explain the different files in the releases?

TurkeyMan, 2022-07-10:
Also, only 3 out of 5 architectures seem to have binary release builds?

TurkeyMan, 2022-07-10:
Also, in the UART flashing instructions, you say "connect TX1, RX1 to the PC", but I'm not sure should I connect the TX on the PC serial to TX1 on the chip, or to RX1? Ie, should they cross-connect, so TX connects to RX, and RX to TX, or should they connect TX to TX, and RX to RX?
Please generally improve these instructions for first-timers :P

TurkeyMan, 2022-07-10:
Also, what is the UART baud rate? That's not written anywhere I can see.

TurkeyMan, 2022-07-10:
One more detail, I've been trying to power the device with 3.3V from the CP2102 USB/UART, but apparently it can't supply enough power... can you please add some information on how to setup the flash environment? How should I power the device while flashing? I don't have a 3.3v DV power supply floating around, and I have no idea where to get one. I thought getting 3.3V from the UART would be fine, but it doesn't work.

LuisThe0ne, 2022-08-17:
The different release names and binary types, are now listed under the releases  as the pull request mentioned here explains.

LuisThe0ne, 2022-08-17:
To connect a microcontroller/chip to your pc (which has rx and tx pins, also called uart interface) you need a ttl or ftdi adapter which plugs in to your pc. To power the chip you can always use the ttl adapter directly if nothing else is connected to it, e.g. if you desoldered the chip, otherwise I would recommend a lab bench power supply or a breadboard power adapter which usually supports 3.3v.

closed 2022-10-02 by openshwprojects