openshwprojects/OpenBK7231T_App #116: random time mqtt disconect bk7231n

Alexxx113, 2022-07-13:
mqtt Mosquitto broker (Haas)
device wifi rele:
firmware OpenBK7231N_1.4.29 work good
firmware OpenBK7231N_1.6.1 - OpenBK7231N_1.7.1
random disconect mqtt
device wifi swith:
firmware OpenBK7231T_1.4.29-OpenBK7231T_1.7.1  work good

Haas log:
`Журнал событий
13 июля 2022 г.
Выключается (off)
18:50:57 - 3 минуты назад
Изменяет состояние на "Недоступно" (disconect)
18:50:46 - 3 минуты назад
Выключается
18:47:39 - 6 минут назад
Изменяет состояние на "Недоступно"
18:47:28 - 6 минут назад
Выключается
18:20:35 - 33 минуты назад
Изменяет состояние на "Недоступно"
18:20:24 - 33 минуты назад
Выключается
17:46:17 - 1 час назад
Изменяет состояние на "Недоступно"
17:46:06 - 1 час назад`

OpenBK log:
`Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MAIN:Time 7511, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7512, free 76264, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 3/38

Info:MAIN:Time 7513, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7514, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7515, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7516, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7517, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7518, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7519, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7520, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MAIN:Time 7521, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7522, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7523, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7524, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7525, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7526, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7527, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7528, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7529, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MQTT:Publishing IP = 5.100.13.144 

Info:MAIN:Forced channel publish! Publishing val 0 with 0 

Info:MQTT:Publishing 0 = 0 

Info:MAIN:Time 7530, free 89704, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MAIN:Time 7531, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7532, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7533, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7534, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7535, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7536, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7537, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7538, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7539, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7540, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MQTT:mqtt_userName user
mqtt_pass passwd
mqtt_clientID shkaf
mqtt_host 5.100.13.248:1883

Info:MAIN:Time 7541, free 89744, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MQTT:mqtt_connection_cb: Successfully connected

Info:MQTT:mqtt_subscribed to shkaf/+/set

Info:MQTT:mqtt_subscribed to cmnd/shkaf/+

Info:MQTT:MQTT client "shkaf" request cb: err 0

Info:MQTT:MQTT client "shkaf" request cb: err 0

Info:MQTT:Publishing IP = 5.100.13.144 

Info:MAIN:Forced channel publish! Publishing val 0 with 0 

Info:MQTT:Publishing 0 = 0 

Info:MAIN:Time 7542, free 89728, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Forced channel publish! Publishing val 1 with 0 

Info:MQTT:Publishing 1 = 0 

Info:MAIN:Time 7543, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7544, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MAIN:Time 7511, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7512, free 76264, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 3/38

Info:MAIN:Time 7513, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7514, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7515, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7516, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7517, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7518, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7519, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7520, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MAIN:Time 7521, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7522, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7523, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7524, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7525, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7526, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7527, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7528, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7529, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MQTT:Publishing IP = 5.100.13.144 

Info:MAIN:Forced channel publish! Publishing val 0 with 0 

Info:MQTT:Publishing 0 = 0 

Info:MAIN:Time 7530, free 89704, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MAIN:Time 7531, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7532, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7533, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7534, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7535, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7536, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7537, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7538, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7539, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7540, free 89952, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:GEN:dhcp=0 ip=5.100.13.144 gate=5.100.13.1 mask=255.255.255.0 mac=c8:47:8c:10:60:6b 

Info:MQTT:mqtt_userName user
mqtt_pass passwd
mqtt_clientID shkaf
mqtt_host 5.100.13.248:1883

Info:MAIN:Time 7541, free 89744, MQTT 0, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MQTT:mqtt_connection_cb: Successfully connected

Info:MQTT:mqtt_subscribed to shkaf/+/set

Info:MQTT:mqtt_subscribed to cmnd/shkaf/+

Info:MQTT:MQTT client "shkaf" request cb: err 0

Info:MQTT:MQTT client "shkaf" request cb: err 0

Info:MQTT:Publishing IP = 5.100.13.144 

Info:MAIN:Forced channel publish! Publishing val 0 with 0 

Info:MQTT:Publishing 0 = 0 

Info:MAIN:Time 7542, free 89728, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Forced channel publish! Publishing val 1 with 0 

Info:MQTT:Publishing 1 = 0 

Info:MAIN:Time 7543, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38

Info:MAIN:Time 7544, free 89952, MQTT 1, bWifi 1, secondsWithNoPing 1, socks 2/38`

openshwprojects, 2022-07-14:
Do you have enabled "Flag 2 -[MQTT] Broadcast self state every minute" in the General settings, and if yes, can you try disabling it?

Alexxx113, 2022-07-15:
Thanks !!!
device wifi rele:
firmware OpenBK7231N_1.6.1 - OpenBK7231N_1.7.1
 not correct works  : enabled "Flag 2 -[MQTT] Broadcast self state every minute"

 **OpenBK7231N MQTT Broadcast - broken**

device wifi swith:
firmware OpenBK7231T_1.4.29-OpenBK7231T_1.7.1
 work good: enabled "Flag 2 -[MQTT] Broadcast self state every minute"

 **OpenBK7231N MQTT Broadcast -work good**

openshwprojects, 2022-07-15:
I think I just experienced the same issue with "Flag 2 -[MQTT] Broadcast self state every minute". My device also stopped responding.

Is device OK "Flag 2 -[MQTT] Broadcast self state every minute"  DISABLED?

Anyway, I am working on it, try checking later today after update and please report back here if it's fixed, ok?

I really need help with testing that.

Alexxx113, 2022-07-15:
OK I'll test it.
It will take time.
Please, add to debag more info mqtt

Thanks !!!

openshwprojects, 2022-07-15:
I improved MQTT stability a bit today but in general - "Flag 2 -[MQTT] Broadcast self state every minute" is NOT reliable now, don't use it

Alexxx113, 2022-07-15:
ok 
I checked OpenBK7231N_1.9.0.rbl
enabled "Flag 2 -[MQTT] Broadcast self state every minute"
the device has already frozen once

I checked OpenBK7231ZT_1.9.0.rbl
enabled "Flag 2 -[MQTT] Broadcast self state every minute"
while the work   ))))

Alexxx113, 2022-07-17:
> I improved MQTT stability a bit today but in general - "Flag 2 -[MQTT] Broadcast self state every minute" is NOT reliable now, don't use it

it is possible to fix it ?
later  ?

Can I help  ?
logs 
test

I made 3 instructions
maybe you need...
https://github.com/Alexxx113/Tuya-switch-WB3S
https://github.com/Alexxx113/Tuya-switch-WB2S-rf-module-
https://github.com/Alexxx113/tuya-rele-WB2S-smart-switch-

Alexxx113, 2022-08-16:
ok
tested:
enabled "Flag 2 -[MQTT] Broadcast self state every minute"
device bk7231n

add 1 channel rele as 0
work 24 hours GOOD!!!!!

add 7 channel rele,ADC as 0,1,2,3,4,5,6
NOT work !
error:
Info:MQTT:mqtt_publish requies 15, has -58715
Info:MQTT:mqtt_output_check_space ERR_MEM while needed 15
Info:MQTT:Publish err: ERR_MEM aka -1

you send Broadcast self state every minute as "while"
can you add pause in while
or 
send as 
mqtt-device-status 
Merge pull request from iprak/mqtt-device-status 

you add send status every 30 seconds

in firmware [1.12.17]

https://github.com/openshwprojects/OpenBK7231T_App/commit/31af7bb69fda82a213af86005de41dda4dc3a315#diff-2e5db50a11df671d296f69c971beab3510e35fdeba10d1d82eb2984f022d1755

**please**  add pause in while send channel "Flag 2 -[MQTT] Broadcast self state every minute"

I do not know "C" language
))))

bugalaws, 2022-08-29:
I got the same error, in a bk7231n the flag 2 is disabled but i got disconnect e ervery minute.. the wifi is stable ok.. 
[image]

bugalaws, 2022-08-30:
> I got the same error, in a bk7231n the flag 2 is disabled but i got mqtt disconnect e ervery minute.. the wifi is stable ok..
> [image]

One more problem, each time mqqt reconnect it turn of my light if it is ON, i’m using the button setup as tglchanontgl …

bugalaws, 2022-09-01:
Log to help :

Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

Info:MQTT:mqtt_userName usuario
mqtt_pass senha
mqtt_clientID LuzCorredorCasa
mqtt_host 192.168.0.20:1883

Info:MQTT:mqtt_connection_cb: Successfully connected

Info:MQTT:mqtt_subscribed to LuzCorredorCasa/+/set

Info:MQTT:mqtt_subscribed to cmnd/LuzCorredorCasa/+

Info:MQTT:mqtt_publish requies 35, has 1024
Info:MQTT:MQTT client "LuzCorredorCasa" request cb: err 0

Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic LuzCorredorCasa/1/set

Info:MQTT:MQTT in topic LuzCorredorCasa/1/set
Info:MQTT:channelSet topic 4210268
Info:MQTT:channelSet part topic 1/set
Info:MQTT:channelSet channel 1
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 1

Info:MQTT:MQTT client "LuzCorredorCasa" request cb: err 0

Info:MQTT:Publishing sockets = 2 

Info:MQTT:mqtt_publish requies 28, has 1024
Info:MQTT:[g_bPublishAllStatesNow] item -5 result 0

Info:MQTT:Publishing rssi = -43 

Info:MQTT:mqtt_publish requies 27, has 1024
Info:MQTT:[g_bPublishAllStatesNow] item -4 result 0

Info:MQTT:Publishing uptime = 1337 

Info:MQTT:mqtt_publish requies 30, has 1024
Info:MQTT:[g_bPublishAllStatesNow] item -3 result 0

Info:MQTT:Publishing freeheap = 85688 

Info:MQTT:mqtt_publish requies 33, has 1024
Info:MQTT:[g_bPublishAllStatesNow] item -2 result 0

Info:MQTT:Publishing ip = 192.168.0.166 

Info:MQTT:mqtt_publish requies 35, has 1024
Info:MQTT:[g_bPublishAllStatesNow] item -1 result 0

Info:MQTT:Publishing 1 = 1 

Info:MQTT:mqtt_publish requies 26, has 1024
Info:MQTT:mqtt_connection_cb: Disconnected, reason: 256

bugalaws, 2022-09-01:
I’ve updated to today update but the problem stil hapens

openshwprojects, 2022-11-04:
@bugalaws we have recently migrated to new LWIP, thanks to the great work of @valeklubomir (he did update LWIP for both platforms and also added much, much features), the stability issue should not be there any more. Please open new issue if anything bad happens.

closed 2022-11-04 by openshwprojects