openshwprojects/OpenBK7231T_App #117: OTA flashing of original tuya devices

beaune33, 2022-07-14:
I am wondering if there is any chance to flash tuya devices with openBK over-the-air. My device is a moodlight, which I can see no chance to open without destroying: https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/device-pictures/3004154.jpeg. As far as I am informed the device contains CB3S chip.

My idea is to make use of the following project: https://github.com/tuya-cloudcutter/tuya-cloudcutter:

> “This repository contains the toolchain to exploit a wireless vulnerability that can jailbreak some of the latest smart devices built with the bk7231 chipset under various brand names by Tuya. The vulnerability as well as the exploitation tooling were identified and created by [Khaled Nassar](https://twitter.com/kmhnassar) and [Tom Clement](https://twitter.com/Tom_Clement) with support from [Jilles Groenendijk](https://twitter.com/jilles_com). Our tool disconnects Tuya devices from the cloud, allowing them to run completely locally. Additionally, it can be used to flash custom firmware to devices over-the-air.”

So,

- The device chip seems to be supported by openBK.
- The device should be able to be flashed ota, as it is mentioned in the supported device list of tuya-cloudcutter project, and the device is older that the bug-fixed firmware.

The remaining questions are:

- How to combine ota flash of tuya-cloutcutter with openBK firmware?
- And: is the device supported by openBK at all?

Maybe this could also be an interesting feature for other devices than mine which are difficult to open. I could try it with my device if anybody could describe what to do in steps. Unfortunately I am not deep enough in this all to figure out myself.

openshwprojects, 2022-07-15:
UG files are used for tuya-cloudcutter. 
Regarding the device, I have no idea what's inside, but if it's a simple PWM LED, then it's already supported, you would just need to guess pins (you can do it by brute force, just try each PWM combination, adjust PWM value in WWW panel and see if light changes).
It should be easy to get it up and working

beaune33, 2022-07-18:
Sounds good… As I can see from https://templates.blakadder.com/lsc_smart_connect_3004154.html it seems to be a normal PWM LED with the following GPIO configuration:

```
R:	GPIO 4
G:	GPIO 12
B:	GPIO 14
CW:	GPIO 5
WW:	GPIO 13
```

There also seems to be a button connected to GPIO 16, but as I understand the information from the link above it can only be used, if an additional resistor is added. So, if I cannot open the device I cannot use it probably.

Just to be sure how to proceed:

- I should try to flash OpenBK7231N_UG_1.10.4.bin to the device, right?
- Unfortunately in tuya-cloudcutter the flashing is not documented in detail. Probably I have to follow this steps:

1. Detach device from tuya cloud as explained.
2. Call `run_flash.sh [wifi adapter name] [device profile] [firmware]`, where
```
Wifi adapter = wlan0
Device profile = 3004154-970743
Firmware = <path>/OpenBK7231N_UG_1.10.4.bin
```

Did you ever try tuya cloudcutter and have any experiences with it?

beaune33, 2022-07-22:
You are right...

After some difficulties setting up a PI for tuya-cloudcutter and unterstandig the expected syntax of the flash script I finally managed to cut my device from tuya cloud and load OpenBeken as a custom firmware sucessfully. Guessing the GPIO I encountered:
```
R:	GPIO 8
G:	GPIO 24
B:	GPIO 26
CW:	GPIO 7
WW:	GPIO 6
```
So, now I am already able to control the device by using the web interface.

I tried using MQTT, and I also suceeded to change the color (e.g. yellow value) by publishing `obk06F6BD7C/2/set` as MQTT topic and a value 0..100 as payload in MQTT Explorer.

What I did not manage to do is the opposite: How to perform explicite reading of a value? As I understood I have to use a similar topic `obk06F6BD7C/2/get`. But what is the payload I have to publish? I didn't manage to get it work in MQTT Explorer till now. Are there any examples which MQTT topics and payloads are defined for LED PWM devices?

Another question:
The original tuya firmware supported a bunch of scenes. Is there a comparable mechanism how I can implement scenes in OpenBeken?

AndyRPH, 2022-07-30:
> You are right...
>
> After some difficulties setting up a PI for tuya-cloudcutter and unterstandig the expected syntax of the flash script I finally managed to cut my device from tuya cloud and load [...]

Were you able to flash this OTA without opening the device up at all?

beaune33, 2022-07-30:
Yes, I never opened the device and even do not know how to open. I flashed it completely OTA. 

The point was to figure out, how tuya-cloutcutter and especially the flash command works. Finally I encountered, that a pi necessarily must be set to english, not german, as one functional result inside the scripts is language dependant, or the script must be changed individually. As flashing command I used:
`./run_flash.sh -r wlan0 LSC/3004154-970743 /work/custom-firmware/OpenBK7231N_UG_1.10.4.bin`

Actually I started integrating it into fhem by using MQTT.

What is still not clear for me is the exact behavior of the openbeken MQTT interface: which topics are provided and how to use them. I found out something just by trying, but I really think there is a lack of documentation. or some examples how to use it e.g. with MQTT Explorer.

I also still don't understand if and how I could implement lightning scenes like sunrise etc. Can I advise the device to fulfil something like a lightning program? By using MQTT?

openshwprojects, 2023-02-22:
Many months has passed and many people managed to flash BK devices via cloudcutter well. I think it's safe to close this issue now.

closed 2023-02-22 by openshwprojects