openshwprojects/OpenBK7231T_App #120: Different WiFi reconnect system

Avitii, 2022-07-15:
My switch restarts every 5-10 minutes because it is in a place where WiFi coverage is poor and it just loses it. This is annoying because the light then turns off. Please make a less intrusive/aggressive system for reconnecting to wifi. I want the switch itself to always be available to manual use.

openshwprojects, 2022-07-15:
1. how do you know that it restarts because of poor wifi? maybe there is something else wrong, do you use MQTT?
2. if it restarts, maybe try using "startup state" of relay channel as -1 so it remembers the last light state before power loss?

Avitii, 2022-07-15:
I tested it on another switch, turning off the router = switch restart, I also checked without mqtt. At the moment, startup state is probably the best and only solution which I found, but not ideal because for a fraction of second the light turns off anyway.

openshwprojects, 2022-07-15:
I will look into it. Or maybe, do you happen to have a device with UART debug log connected (not the network one, of course) and can capture the full log?

If not, wait a day or so, I will try to disconnect my router and watch what happens to a device on UART. It's just I can't do it at most of the time because I'd disconnect all family members from the internet.

Avitii, 2022-07-17:
Unfortunately, I don't have option to read logs. I can wait, for now, I changed switch mode to AP.

Avitii, 2022-11-02:
Then how? Is there a chance to change the wifi reconnection system? Restarting the device if the WiFi network is lost is the dumbest solution ever.

valeklubomir, 2022-11-02:
It is not designed to restart. Restart must be critical bug which forces restart of device. I was testing with restarting wifi router and it just reconnected. I see your previous comment was at 17 Jul. Me self been working here on project since just one month and trying to improve Wifi and MQTT communication. You could try newer version. Which platform do you have?
There have been lately some releases with issues. But some are stable. First the platform?