openshwprojects/OpenBK7231T_App #121: add ADC Temp

Alexxx113, 2022-07-16:
please add

NTC thermistor a B value of 3950 100k
resistor 20k
scheme:
[https://github.com/Alexxx113/hlam/blob/main/IMG_20220716_015555.jpg](url)

tasmota name ADC TEMP

work haas
config:

sensor:
  - platform: mqtt
    state_topic: "koridor/5/get"
    name: "mqtt temp sens koridor"
    qos: 1
    device_class: power
    availability_topic: "koridor/connected"
    #device_class: temperature
    #state_class: measurement
    #last_reset_value_template: "00"

template:
  sensor:
    - name: Temperatura NTC
      unique_id: temperatura_koridor 
      device_class: temperature
      state: >
        {% set Vo = states('sensor.mqtt_temp_sens_koridor')|float %}
        {% set c1 = 1.009249522e-03 %}
        {% set c2 = 2.378405444e-04 %}
        {% set c3 = 2.019202697e-07 %}
        {% set R1 = 100000 %}
        {% set R2 = R1 * 1023 / ( Vo - 1 ) %}
        {% set logR2 = log(R2,10) %}
        {% set T = ( 1 / ( c1 + c2 * logR2 + c3 * logR2**3 ) - 273.15 )  / 5 -11.3 %}
        {{ T |round(1) }}
      unit_of_measurement: '°C'

found an example
I have selected the parameters , under the current temperature )))

but this is not right
this is without calibration table

openshwprojects, 2022-07-16:
Hey, how may I help you?

ADC pin should be already working and storing readout correctly in channel. Then you can receive channel in Home Assistant and process it however you like.

Are you referring to "ADCParam" calibration in Tasmota?

Alexxx113, 2022-07-16:
> Are you referring to "ADCParam" calibration in Tasmota?
Yes!
it will be good!
it's easier to configure

when will there be free time.

thanks!

Tasmota
Comands
https://tasmota.github.io/docs/ADC/#commands

Alexxx113, 2022-07-28:
thanks!

normal home assistant formula

 - name: Temperatura NTC kabinet
      unique_id: temperature_kabinet
      device_class: temperature 
      state: >
        {% set Vo = states('sensor.mqtt_temp_sens')|float %}
        {% set res = 200000 %}
        {% set term  = 100000 %}
        {% set nom_t  = 27 %}
        {% set adc  = 3584 %}
        {% set B = 3950 %}

        {% set tr  =  res / ((adc / Vo) - 1)  %}
        {% set steinhart = term /  tr%}
        {% set steinhart1 = log(steinhart) %}
        {% set steinhart2 = steinhart1 / B %}
        {% set steinhart3 = steinhart2 + (1.0 / (nom_t + 273.15))%}
        {% set steinhart4 = 1.0 / steinhart3  %}
        {% set steinhart5 = steinhart4 - 273.15%}

        {{ steinhart5  |round(2) }}
      unit_of_measurement: '°C'

closed 2022-07-28 by Alexxx113