openshwprojects/OpenBK7231T_App #122: Compatibility with RTL8710BX

mihsu81, 2022-07-16:
First of all thanks a lot for your amazing work.
Since it's not possible to start a discussion I had to open an issue.
Do you still have plans for the RTL8710BN, since I don't see releases for it and there only seems to be a demo available?
And would it be compatible with a RTL8710BX?
I have an EZVIZ T31 smart plug which I would like to convert.
Thanks in advance.

openshwprojects, 2022-07-16:
It is planned and almost done. Do you need that support? I can adjust my roadmap if that's needed.

mihsu81, 2022-07-16:
There's no rush. I'd be more than happy to test if you have beta binaries.

mihsu81, 2022-09-25:
Additional info about the implementation of the EZVIZ T31 (RTL8710BX) in libretuya, in case it helps your progress.
https://github.com/kuba2k2/libretuya/issues/22
https://github.com/kuba2k2/libretuya/issues/23
https://github.com/kuba2k2/libretuya/issues/24

brijins, 2022-11-28:
Hi,

When we can expect a binary for testing?  an approximate date would be helpful, just curious. 

Thanks!

divadiow, 2025-01-28:
RTL8710B test builds https://github.com/openshwprojects/OpenBK7231T_App/pull/1501