openshwprojects/OpenBK7231T_App #125: Devices hosed after OTA update to 1.10.4

iprak, 2022-07-22:
I updated 3 similar OpenBK7231T plugs from 1.4.16 to 1.10.4 (OpenBK7231T_1.10.4.rbl) and they all stopped working. 

The current version is now 1.10.7 but I can't tell from the changes if any fix was introduced in this regard.

openshwprojects, 2022-07-22:
This sounds like a serious issue.

Did you use N chip version on T plug? It would hose them totally.

Is the safe mode working?

Can you open one device and provide log to check what goes wrong?

iprak, 2022-07-25:
The devices were dead. I did download the hosed firmware before flashing the new one but the extracted file size does not match T or N firmware bin files. 

I think I must have flashed the wrong firmware. 

Would it be possible to show the firmware type (T/N) on config page in addition to the vesion?

openshwprojects, 2022-07-26:
Would that be enough? Added in latest version (clear browser cache!)
[image]

iprak, 2022-07-26:
That should help. I can see a check in OTA someday update itself pointing this mismatch.

Is that in 1.12.0 ? I am seeing empty chipset

[image]

There was no such data piece in the response from server:
[image]

openshwprojects, 2022-07-28:
This is because the source code of javascript panel is hosted online and the chipset request get data is in flash file, so all older flashes will return empty chipest.

Just update the flash if you need. Don't update if that's not required, it's okay to have an empty field here

closed 2022-08-02 by iprak