openshwprojects/OpenBK7231T_App #128: Is there a way to disable multi touch for wall touch switch

Stef-90, 2022-07-23:
Like tasmota option13 
Currently touch switch have a delay when pressed and only turn on/off light when released.
It's same thing with esp based switches and tasmota before enabling that option 13, which basically tells it to use only single touch input and not to wait for double tap or long tap.

Is there a way to do same thing with bk based switches?

openshwprojects, 2022-07-24:
That's a very nice idea, I have just added it for you, can you check if that's the expected behaviour?
https://github.com/openshwprojects/OpenBK7231T_App/commit/3dc86ac6d472f73b3e8ee01b3cd3dbfdba45bf44
PS: You could also use scripting and event handlers to get similiar results

NOTE: I don't have any touch B7231 buttons at hand to check, so please verify the commit works as expected!

Stef-90, 2022-07-26:
Wow you're quick. I have been busy with work.
Will review the code and and flash today.

Stef-90, 2022-07-26:
Update. flashed and it works perfectly. PR includes setting instant touch for this type of device by default. 
Is there any other way to enable disable custom flags?

openshwprojects, 2022-07-26:
"Is there any other way to enable disable custom flags?"
I can add a console command for that

Stef-90, 2022-07-26:
This works for me, and I can always update templates with any new device that I get, but might be beneficial for someone who have different pinout touch switch, so they can configure their switch manually, and enable it via console.

closed 2022-07-26 by Stef-90