openshwprojects/OpenBK7231T_App #131: Consider migrating to PlatformIO

anogma, 2022-07-25:
I know you've been busy implementing and revising contributions, but I think it may be interesting if the project could be migrated to PlatformIO. It would take some time but it may help to attract new contributors.

iprak, 2022-08-10:
I am not in-depth familiar with PlatformIO and could not figure out if a single instance could target multiple chips. Do you know if that was possible or if separate instances would have to be started? I decided to use Docker instead and use it to verify build on 3 basic chips- https://github.com/iprak/OpenBK7231T_Dev

rpavlik, 2022-11-24:
I tried this a while ago but had trouble getting the Beken sdk, etc working in it. (It's an old ARM9 architecture, not actually cortex M4 like tuya says, and there are binary blobs for WiFi support, etc that seem to depend pretty closely on the shipped compiler version.)

You definitely could run this in platform io, the Marlin 3d printer firmware uses it and runs on lots of chips, hardest part is just getting the platform set up.