openshwprojects/OpenBK7231T_App #136: CB2S cannot flash, "Cannot get bus."

Suxsem, 2022-07-26:
Hello guys,
I'm trying to flash OpenBK7231N_QIO_1.12.0.bin on my CB2S

[image]

I'm using a ft232rl USB<-> TTL adapter. I have connected:
RX -> TX
TX -> RX
GND -> GND

I didn't soldered the Vcc pin because I'm powering the module from the mains voltage (I'm using un unplugged notebook to avoid current loops).

So these are my steps:
1) `python uartprogram OpenBK7231N_QIO_1.12.0.bin --unprotect -d com6 -w --startaddr 0x0`
2) As soon as I get the "programm...." I connect the module to the mains voltage
I see the led blinking but I always get "Cannot get bus.". Can you help me please?

Thank you!

death2all110, 2022-07-26:
Look for a pin labeled CEN. https://developer.tuya.com/en/docs/iot/cb2s-module-datasheet?id=Kafgfsa2aaypq

Solder a wire to it, and leave it floating and once you have the module powered, jump it to ground. Data should then start flowing.

Suxsem, 2022-07-26:
@death2all110 tried with no luck...
I also tried to power the module with the UART adapter 3.3v without any luck...
i'm stuck

death2all110, 2022-07-26:
Which device is this? Also, just to confirm, you are soldered to TX1 and RX1 correct? TX2 and RX2 will not work, those are just for logging output.

Suxsem, 2022-07-26:
@death2all110 moes MS-150B (dual channel dimmer).
Yes TX1 and RX1

openshwprojects, 2022-07-26:
I still WOULD NOT try to mess with mains, it's very dangerous! 

Maybe it's a TuyaMCU device?

Stef-90, 2022-07-26:
IF it's TuyaMCU it uses TX1 and RX1 to communicate with TuyaMCU, so you need to unsolder these two pins before flashing.
Other options that I had success with sometimes is to just unsolder vcc 3v from bekken board, then power it from there, that way tuyaMcu doesnt get powered and it doesn't interfere with tx/rx1.
But best bet is to unsolder both Rx1 and Tx1 it worked for me 10/10

l3ztum, 2022-07-28:
my device had rx soldered to a switch. there was a little external pulldown resistor soldered to the cb2s module. i couldn't flash as long as the mentioned resistor was in place. maybe your board is wired up in a similar way.

openshwprojects, 2022-11-04:
Yes, you should check for capacitors and pull down resistors, and for TuyaMCU on TX/RX lines.

closed 2022-11-04 by openshwprojects