openshwprojects/OpenBK7231T_App #137: Home assistant recognize new Tuya device (all devices are flashed)

Stef-90, 2022-07-27:
Is there something that is still running in background that's trying to connect to tuya, even though they are custom flashed?
Might be worth investigating and if that's the case disabling it.

openshwprojects, 2022-07-27:
I don't think there is anything running in the background. user_main_entry directly creates the main app thread, nothing else. I'd say that it recognizes by MAC address.

btsimonh, 2022-07-27:
You could probably dig into the tuya integration code in HA to find out what it's doing.
Also, it may be logging into your tuya cloud account, and not seeing the local device at all, just that it once was registered?  That's most likely as I don't think the Tuya integration could work against local devices anyway.... not many are enabled for local comms, it all goes through their cloud.

Stef-90, 2022-07-28:
I don't have tuya cloud, but scan by Mac address sounds like it could be the answer. Will dig into HA code to confirm

ardenking, 2022-08-19:
did you test them on the Tuya cloud before flashing ?

konradmb, 2022-10-31:
It is scanning by MAC: https://github.com/home-assistant/core/blob/e63616987848772c10c2e5fbb607278e80b6bfc5/homeassistant/components/tuya/manifest.json#L10
Compare: https://developers.home-assistant.io/docs/creating_integration_manifest/#dhcp

To fix:

- Change first part of MAC in OpenBK config (for example increase one digit - 105A17 becomes 205A17)
- Reboot whole OS (restarting only HA still shows the "New integration discovered" message)

20:5A:17 is unregistered in MAC DB https://macaddress.io/mac-address-lookup/eJ2aaqXY2j