openshwprojects/OpenBK7231T_App #138: Generate Home Assistant cfg using deprecated configuration format

dstreefkerk, 2022-07-27:
Hi there!

I used the "Generate Home Assistant cfg" feature to create the config for a 3-gang switch in Home Assistant, and later noticed in the Home Assistant logs that there was warning about a deprecated switch configuration.

To quote [this page](https://www.home-assistant.io/integrations/switch.mqtt/#new_format): 

> "The configuration format of manual configured MQTT items has changed. The old format that places configurations under the switch platform key should no longer be used and is deprecated."

Here's the config generated for my device by OpenBK:
```
switch:
  - platform: mqtt
    name: "familyroom-lightswitch 1"
    state_topic: "familyroom-lightswitch/1/get"
    command_topic: "familyroom-lightswitch/1/set"
    qos: 1
    payload_on: 0
    payload_off: 1
    retain: true
    availability_topic: "familyroom-lightswitch/connected"
  - platform: mqtt
    name: "familyroom-lightswitch 2"
    state_topic: "familyroom-lightswitch/2/get"
    command_topic: "familyroom-lightswitch/2/set"
    qos: 1
    payload_on: 0
    payload_off: 1
    retain: true
    availability_topic: "familyroom-lightswitch/connected"
  - platform: mqtt
    name: "familyroom-lightswitch 3"
    state_topic: "familyroom-lightswitch/3/get"
    command_topic: "familyroom-lightswitch/3/set"
    qos: 1
    payload_on: 0
    payload_off: 1
    retain: true
    availability_topic: "familyroom-lightswitch/connected"
```

And here's my revised config, following the new practice for MQTT switches:
```
mqtt:
  switch:
    - unique_id: familyroom_light_switch_1
      name: "Family Room Light Switch 1"
      state_topic: "familyroom-lightswitch/1/get"
      command_topic: "familyroom-lightswitch/1/set"
      availability:
        - topic: "familyroom-lightswitch/connected"
      payload_on: 1
      payload_off: 0
      qos: 1
      retain: true
    - unique_id: familyroom_light_switch_2
      name: "Family Room Light Switch 2"
      state_topic: "familyroom-lightswitch/2/get"
      command_topic: "familyroom-lightswitch/2/set"
      availability:
        - topic: "familyroom-lightswitch/connected"
      payload_on: 1
      payload_off: 0
      qos: 1
      retain: true
    - unique_id: familyroom_light_switch_3
      name: "Family Room Light Switch 3"
      state_topic: "familyroom-lightswitch/3/get"
      command_topic: "familyroom-lightswitch/3/set"
      availability:
        - topic: "familyroom-lightswitch/connected"
      payload_on: 1
      payload_off: 0
      qos: 1
      retain: true
```

Also note that my new config has payload_on and payload_off values swapped around. The default generated config was showing switches as "off" when they were actually "on"

openshwprojects, 2022-08-02:
Thanks, it should be fixed in just merged pull requested by @iprak  
https://github.com/openshwprojects/OpenBK7231T_App/commit/b7d185eb584d0b6b659d6231c0352b70e015ae60
great work, I had a busy last week so was unable to do it myself.

Please check if it's ok for you now.

iprak, 2022-10-20:
@dstreefkerk This should no longer be an issue, please check.