openshwprojects/OpenBK7231T_App #139: BTN_tgl_all send wrong MQTT value

thanasiskalos, 2022-07-30:
Hello
BK
I have managed to flash a power strip with BK7231 successfully . 
I noticed though that upon selecting the btn_tgl_all on first press sends MQTT values on each switch/relay as 0 on second press sends for MQTT on each Switch/ Relay value 255 which it is not know to MQTT server and Home assistant. A value of 1 should be send instead.

Please have a look.

openshwprojects, 2022-08-02:
Thanks, I will look into it this week

openshwprojects, 2022-08-03:
Hey, can you check?
https://github.com/openshwprojects/OpenBK7231T_App/commit/82cd86a65a59144c6aa1cd39a4b48000248f7dec
it now sets relays to 1 and pwms to 100 (which are correct values)

thanasiskalos, 2022-08-03:
Hello

Seems to work fine now.

Keep up the good work. It is amazing what you do and the commitment you are having.

Hope your work gets acknowledged.

Regards

closed 2022-08-03 by thanasiskalos