openshwprojects/OpenBK7231T_App #142: Feature Request: Add Support for BP5758 I2C LED driver

death2all110, 2022-08-03:
The BP5758 is an I2C LED driver, but is not strict I2C. Support for this driver would need added before I can use this firmware with my bulbs. I have very limited knowledge of C code so I can't really do a pull request. However, if someone would like to tackle this project, I have located a datasheet and some code others have written, one for ESPHome and another for Tasmota.

Datasheet: https://raw.githubusercontent.com/dbuezas/esphome-bp5758/main/BP5758D_EN_DS_Rev.1.1.pdf
ESPHome Driver: https://github.com/dbuezas/esphome-bp5758
Tasmota: https://github.com/arendst/Tasmota/pull/15713/files#diff-d975994017d366f063cb8fdcd7bfaf7115ad8c879664b2ea85b0b7ff2093c925
Tuya's Developer page: https://developer.tuya.com/en/docs/iot-device-dev/driver_i2c_bp5858d?id=Kb7bciwq50zfc

I'm happy to test things if anyone can get this driver implemented.

openshwprojects, 2022-08-03:
Hey, thanks, thank you also for submitting that device on Elektroda site.
I will try tomorrow to port this driver, remind me if I forget.
I will really need your help with testing, as I don't have this device.

death2all110, 2022-08-03:
> 

Sounds good. This won't be a problem. I'm happy to help test.

openshwprojects, 2022-08-04:
I don't know, you might try this alraedy:
https://github.com/openshwprojects/OpenBK7231T_App/commit/a81ac2f24baae2d921fe1767268ba350d9a68f4c
set pin roles, start driver (startDriver BP5758D;or it will start automatically after restart if pins are set) and use:
BP5758D_RGBCW FF00000000
this is also linked to LED driver now, but I am not able to test it as I don't have that device.

openshwprojects, 2022-08-04:
5 times on and off

death2all110, 2022-08-04:
Thanks, I was able to get back to the OpenBK7231N_XXXXX ssid and connect 

I removed the BP5758D pin roles and saved and saved the wifi config again and I'm back on my router's SSID. 
Assigned the pin roles again, issued the command to start the driver, and it went offline again.

openshwprojects, 2022-08-04:
Oh no, I super screwed up, no wonder it crashed.

Can you check again, the commit I made right now?

death2all110, 2022-08-04:
Sure. I'm not seeing a new OTA file, so will I need to build myself with this latest commit?

openshwprojects, 2022-08-04:
I think it's up, it had to build online

death2all110, 2022-08-04:
Yes, I see it now, I guess I was a bit too quick. Lol.

death2all110, 2022-08-04:
Unfortunately, it still seems to be crashing on me...

openshwprojects, 2022-08-04:
and the new version?

death2all110, 2022-08-04:
> and the new version?

Yeah, 1.12.11 is what is crashing for me still. Would it help if I opened the bulb and got the logs from the uart2

openshwprojects, 2022-08-04:
I meant 1.12.12 or 1.12.13

death2all110, 2022-08-04:
> EDIT: try 1.12.14 when it comes up

Will do. Not home at the moment but will check as soon as possible. Thanks again! You're a rockstar!

death2all110, 2022-08-05:
Success!!

You rock!

openshwprojects, 2022-08-05:
It's a very similiar protocol but different addresses and data kinds, for example brightness values are 0-1023 and not 0-255.

It would work at the first try if I had more time and focus to check for silly mistakes like wrong function called ("Write" instead of "WriteByte")

do you know any other drivers that should be added?

PS: You can also post in your Elektroda topic that driver is supported now

openshwprojects, 2022-08-07:
Thank you for the testing

closed 2022-08-07 by openshwprojects

syph3rd, 2022-09-06:
Hello, 

I am working with an Australian RGBCW Globe with a CB2L and BP5758 Chip, I have tried the firmware and can't get comms or logging on the I2C for comms to the chip.