openshwprojects/OpenBK7231T_App #143: Device stopps communication after MQTT publish

beaune33, 2022-08-03:
I am using LSC Smart LED Mood Light (3004154) with BK7231N inside. When I go to the web-based command tool and enter `led_enableAll 1` or `led_enableAll 0` I can switch my device sucessfully on and off.

Trying the same with MQTT using topic `cmnd/obk06F6BD7C/led_enableAll` and payload `1` or `0` also suceeds once, but afterwards the device hangs and does no longer answer pings. It needs a hard reset to recover.

This behaviour was watched with firmware 1.10.4, but also 1.12.7.so no changes in the past. What is wrong?

openshwprojects, 2022-08-03:
Can you check if it happens also in earlier revisions?  The ones before 16 Jul 2022?

beaune33, 2022-08-03:
I tried with 1.9.0, and it happenes also.

openshwprojects, 2023-02-22:
I think this old issue was fixed thanks to @valeklubomir  and @btsimonh  efforts, the MQTT is stable now, so I am closing this topic.

closed 2023-02-22 by openshwprojects