openshwprojects/OpenBK7231T_App #144: Switch status in web app does not match config

iprak, 2022-08-03:
I have a BK7231T device with 1.12.9 firmware and am seeing the switch status displayed differently. This was happening even before 1.12.9 . The switch is actually on in this case.

[image]

openshwprojects, 2022-08-07:
I could try looking into it, but as said earlier, the web app is here:
https://github.com/OpenBekenIOT/webapp
and was written by @btsimonh , VUE scripting is not my thing.

If anyone know VUE better, you can look at source here:
https://github.com/OpenBekenIOT/webapp/tree/gh-pages

iprak, 2022-08-07:
Not a big deal. I am looking into the webApp.

iprak, 2022-08-07:
Any ideas how a local website can be used to server content without tripping Cross-origin policy check?

closed 2022-08-10 by iprak