openshwprojects/OpenBK7231T_App #147: [Request] Aubess Mini Switch 16A (BK7231N )

pretoriano80, 2022-08-06:
Hi,i've bought an Aubess Mini Switch from Aliexpress ,but unfortunately it uses BK7231N on a custom PCB ,si i'm struggling to find the correct PINS for flashing OpenBeken firmware .

So far the only pinout infos i've found ,are from the modules that use BK7231N ,but not the chip itself .

It also seems to use TuyaSDK 3.4 which is not yet supported by TinyTuya ,so local control is not possible either.

I will attach the pictures here,so let me know if you have any idea on the pinout of this thing .Thank you !
[image]
[image]
[image]

openshwprojects, 2022-08-06:
No problem, it's very easy. Just remember that first pin is where the dot mark is and then use this diagram:
[image]

Also note where the oscillator is (26MHz), it is using XI and XO pins.

pretoriano80, 2022-08-06:
Thanks ,i will try to figure it out .Which pin should i use for VCC ? (did some tests with the multimeter and P1 seems to be GND,but not really sure on VCC because there's 3.3v on multiple pins).

openshwprojects, 2022-08-06:
What kind of approach are you planning to take with powering it up?

If you have 3.3V LDO regulator, then you connect your 3.3V directly to the 3.3V pin. 3.3V pin on my diagram is simply VCC.

I think I can see the VCC track from LDO regulator to the BK chip with extra 100nF SMD capacitors (maybe also 1uF or 10uF) on your photo.

pretoriano80, 2022-08-06:
The UART dongle i have can provide both 3.3v ond 5v,it has a switch for it.
So far it seems that PCB PAD4 = TX and PAD5=RX.

pretoriano80, 2022-08-06:
I will go with :   
PAD8 -> VCC
PAD1 -> GND
PAD4 -> TX
PAD5 -> RX

pretoriano80, 2022-08-07:
Flashed OpenBeken successfully ! Now i'm trying to figure how to configure (map)  relay,button and led

openshwprojects, 2022-08-07:
That's great! 
If you want to help, please post a short device teardown with the pins marked (which pin is which) in our teardowns section:
https://www.elektroda.com/rtvforum/forum507.html
Preferrably device model name, package photo, shop screenshots, photos from inside and openbeken config. Thanks!

pretoriano80, 2022-08-07:
I managed to configure everything and it works (embedded button,led and external momentary switch ).

I will create and account on elektroda and also update it here on GitHub .

pretoriano80, 2022-08-07:
@openshwprojects  as i said,everything seems to work ,but i'm confused about a strange behavior of the WebUI.
When the relay status is OFF in the WebUI homepage ,it shows as ON in the WebAPP and Homeassistant and viceversa .

I controlled with the multimeter and the Webapp and Homeassistant are showing the correct state .
Any idea why this is happening ?

openshwprojects, 2022-08-08:
You are correct. It seems now it's fixed by @iprak

pretoriano80, 2022-08-08:
I will check again over the weekend ,but with la build from a few days ago it was still wrong .
And as far as I can tell ,the Homeassistant configuration that was generated still had the wrong payload values (payload 0 = ON and payload 1 = OFF).

Modifying the config file manually fixed the issue on the http://device_ip/index  page (which is now in sync with mqtt relay status ),but it's now wrong on the WebApp.

iprak, 2022-08-08:
I pushed the fix last night and I don't the change has yet made to the published webApp. !openshwprojects/obj_webapp/pull/1

openshwprojects, 2022-08-09:
@prak i just realized and fixed that. You pushed to my private fork and the public repo is in openbekeniot ... but should be already fixed, next time please push to that correct repo

iprak, 2022-08-12:
@pretoriano80 You are right, the config HA generated is reversed. What I had addressed was the handling in the webApp. I will fix the incorrectly generated config too.

By the way, I also have a Aubess MiniSwitch with a different board. The BK7231N chip is on a separate board and then pins are very accessible.
[image]
[image]

pretoriano80, 2022-08-12:
Yeah ,it was also reported in the bug report #138 .
Maybe yours is an older Aubess switch ,mine seems that it was made in April 2022.
It's easy to flash now that i know which pins must be soldered and to which pads.It does require sone soldering skillls.

openshwprojects, 2022-08-12:
I had similiar device and had to cut the track in order to flash:
https://www.elektroda.com/rtvforum/topic3895572.html

pretoriano80, 2022-08-15:
Also posted here -> https://www.elektroda.com/rtvforum/viewtopic.php?p=20145042#20145042

openshwprojects, 2022-08-16:
Thanks @pretoriano80 , we will move this to public section after it's been checked by moderator. If you have any other devices, feel free to post them as well.

Also consider doing flash dumps (full 2MB) for Tuya-cloudcutter

pretoriano80, 2022-08-16:
I don't have other tuya wifi  devices (i prefer zigbee ) apart a few Yeelight Bulbs which support Local Connection so no need to mess with the firmware ,i only block internet access to those devices (and also block DNS requests with Adguard ).

pretoriano80, 2022-08-16:
@openshwprojects is there a way to run webapp locally?I couldn't find any documentation about it.

iprak, 2022-08-16:
@pretoriano80  I think I have same setup as yours, all my IOT devices are firewalled.

I was not able to consume webApp from a local website. The webApp code is in `https://github.com/openshwprojects/obj_webapp.git` in the branch `gh-pages`. I was running into cross-site error and btsimonh suggested `checking CORS headers in the App URL serving from the Device code`. Since I configure the devices from an internet accessible machine so I did not bother messing with this any further.

pretoriano80, 2022-08-16:
@iprak it's CORS related indeed --> `access has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.`

pretoriano80, 2022-08-16:
I'm using SWAG container to test this webapp stuff and i use reverse proxy to access most of my services/containers.

iprak, 2022-08-16:
I think you need to add the header `Access-Control-Allow-Origin` with a value of `*` in the nginx? webserver configuration.

pretoriano80, 2022-08-16:
Yes.thats's what i did,but instead of using a wildcard, i used the device's local address.But there's something strange,it works on desktop browsers ,but not on mobile...

openshwprojects, 2022-08-16:
@pretoriano80 I think you can change web app URL in the Config on WWW panel

@iprak  you are pasting incorrect URL here, the URL used officialy is:
https://github.com/OpenBekenIOT/webapp

pretoriano80, 2022-08-17:
@openshwprojects  yes,that's how I've set the local URL,but that alone won't work because for cross-site `Access-Control-Allow-Origin` must be set in the server configuration .

It could be set like @iprak suggested ,with `add_header "Access-Control-Allow-Origin"  *";` ,but that's a wide-open configuration which accepts cross-site from any source so i preferred add_header "`Access-Control-Allow-Origin"  "http://SERVER_NAME OR IP";`

Also ,it's best if the server has a valid SSL certificate,otherwise on mobile browser you will only get a blank page due to the SSL certificate error and to fix that you must open (just once) the webapp page directly (e.g https://server_ip/webapp/ )and accept the security risk,after that the webapp will work directly from the device's WebUI.

I must say that i didn't tested with plain HTTP because on my server port 80 is used by another service so i don't know if it works better or not.

iprak, 2022-08-19:
@pretoriano80 I have a local non-https webapp successfully running on apache with just this extra config in `/etc/apache2/apache2.conf`

```
<Directory /var/www/html>
     Header set Access-Control-Allow-Origin "http://192.168.1.213"
</Directory>
```

pretoriano80, 2022-08-19:
I was pretty sure that with HTTP it's easier because no certificate is involved .
However ,it works with SSL and the trick i explained in the previous message .

iprak, 2022-08-19:
@pretoriano80 So I am curious why/how do you use custom website?. I initially tried it with the intention of not having to access internet but I am usually doing this from a web accessible machine. The webApp also relies on some external resources and so would not work from a locked down machine.

pretoriano80, 2022-08-20:
@iprak usually i like to run the stuff locally if possible .
Are you sure that external resources are used,i'm asking because i can't find any DNS request in Adguard Home query log?

iprak, 2022-08-20:
No you are right. I mistakenly was thinking that `index.html` was loaded but that is only navigated when one visits the webApp directly. Otherwise `startup.js` just loads the files locally.

jkwim, 2023-01-03:
Can somebody explained how to get this going please?

When you look at [https://github.com/openshwprojects/obj_webapp/tree/gh-pages](https://github.com/openshwprojects/obj_webapp/tree/gh-pages) the status says:

This branch is [291 commits behind](https://github.com/openshwprojects/obj_webapp/compare/gh-pages...OpenBekenIOT:webapp:gh-pages) OpenBekenIOT:gh-pages.

Just for the sake of trying I downloaded the ZIP file from above page and set it up to be accessible at a local server using lighttpd.

Direct access to the web server gives me 

`Hello from vue3-sfc-loader v0.8.4
`
Then I updated the URL in the device and tried to access the webapp at [http://<deviceip>/app?](http://<deviceip>/app?) but don't see anything happening.

Questions:
1. Where can I get the WebApp pages in the main branch? (I am assuming that the above gh-pages does not get sync'd with the latest code.)

iprak, 2023-01-03:
You would want to use https://github.com/OpenBekenIOT/webapp .. the actual code is in `gh-pages` branch.