openshwprojects/OpenBK7231T_App #149: CB3S resets when scanning WiFi

ardenking, 2022-08-08:
Thanks for the firmware tested it today with some CB3S I'd removed a few month back

Works fine but if I tell it to scan WiFi it becomes unresponsive and resets, manual entering ssid works fine so have a work around

luke-jr, 2022-09-11:
I couldn't get its default AP to stay active for more than a few seconds. Managed to scan once fine. When I did finally set wifi info, it bootlooped. :/

rojoricardo, 2023-04-23:
> I couldn't get its default AP to stay active for more than a few seconds. Managed to scan once fine. When I did finally set wifi info, it bootlooped. :/

Try having a better power source, these chips tend to be power hungry and sometimes is better to get them back into the original devices power