openshwprojects/OpenBK7231T_App #150: Big pwm brightness jump from 99 to 100

DSchndr, 2022-08-09:
Hi, I flashed a lsc smart led 1400lm (wb2l & KP18055ESPA) with openbeken and there is a giant difference in brightness on both the cold and warm channel when going from 99 (or lower) to 100

Is there a bug with the pwm code?

What happens when the channel is set to 100? 
Does it set the duty cycle directly?

One Idea was to change the pwm freq since the chinese datasheet says "fDim: 540-600-660 HZ", where is it defined?
Option for pwm freq in the webinterface would be nice 

Btw, nice project; thanks ^^

openshwprojects, 2022-08-09:
I have WB3S LED dimmer and have not noticed that...
Here is PWM code:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/hal/bk7231/hal_pins_bk7231.c
100 is no special value, everything should be ok. I have also tested it with LEDs.

Do you have a scope?

Hmmm frequency is constant, but I can add an option for that...

That frequency setting, would you like it to be global or per-PWM?

DSchndr, 2022-08-09:
If 100 are 100% duty cycle i guess the driver (KP18055ESPA) just does current limiting since it is not dimming
I guess 1khz is too fast for some part in that blackbox driver, why would they else specify a recommended freq?

I have a scope buut the lamp is dangerous to handle without an isolation transformer xD

A per channel field would be more flexible and better I guess (if, for example a rgbcw lamp uses different drivers) 

I would have compiled and tested a version with 600hz pwm but have to dig out a linux machine (wsl is incompatible with 32bit compiler binary) :/

openshwprojects, 2022-08-09:
so maybe wait a day or two and I will add this option

DSchndr, 2022-08-17:
That driver ic is wierd, setting the pwm frequency to 650Hz causes the leds to light up at 20 and have the same behavior ("nightlight" to eye melting torch) at 65

(1khz has that extreme brightness perfectly at 100, could be used as a feature)

1010hz seems to work, i have no 1400lm reference here but at least it dims properly from ~19 to 100

On higher frequencies (like 6khz) the lamp starts to light up earlier but you loose brightness

Flashing stock fw and looking at what the ofw does pwm wise would be best i guess

pretoriano80, 2022-08-26:
I can confirm this issue,the jump from 99 to 100 it's huge,both for cold white and warm white .
And there's something off with the way brightness is adjusted (probably related to this issue ),because everything below 99% has pretty low brightness ,there are no smooth steps from WW -> Neutral White -> CW and viceversa .

DSchndr, 2022-08-27:
@pretoriano80 Which driver chip does your lamp use, the same?

pretoriano80, 2022-08-27:
@openshwprojects i did more testing and played a bit in Homeassistant with Brightness and Color Temperature sliders .

Homeassistant Brightness and Color Temperature sliders values : 

Brightness = (min) 0  -> 100 (max)
Color Temperature = (min) 153 -> 500 (max)

1)  B:100  + CT:153/154 =  Cold White at its max brightness (a light sensor that i have reported 95 lux)

2) B:100 + CT:155 = Cold White at very low brightness (8-9 lux reported by the light sensor)

3) B:100 + CT: 300 = (not so) Neutral White at very low brightness (8 lux)

4) B:100 + CT:499 = Warm White at very low brightness (7-8 lux)

5) B:100 + CT:500 = Warm White at full brightness (around 80 lux)

Other tests : 

B:10 +  CW =  Bukb completely OFF
B:10 + WW = Bulb has dimmed light as expected .

pretoriano80, 2022-08-27:
@DSchndr idk if this IC is driving the leds,but couldn't find any info about it.
[image]

DSchndr, 2022-08-27:
@pretoriano80 Yes, same ic
I have a build with different pwm freq, only issue is what i've stated above - its kinda dim and not even near its full power but it works
There must be some magic going on which my european eyes can't see lol

pretoriano80, 2022-08-27:
The light bulb in my case is this -> https://www.ngs.eu/en/smart-home/lighting/gleam-1027c/SMARTHOME/SMT-ILLU-0015/  <-  i have two of them,one flashed with OpenBeken and one that i'm using with Homeassistant and Local-Tuya integration.
If you think there's something that might help from the stock Tuya one ,just let me know

pretoriano80, 2022-08-28:
Digging around the internet i've found more informations :  

https://developer.tuya.com/en/docs/iot/product-hardware-design-considerations?id=K9s9rhj4aqa8e&_source=github#subtitle-7-CW%20control%20(linear)

https://support.tuya.com/en/help/_detail/K9hri6jj01loj

https://developer.tuya.com/en/docs/iot/pwm?id=Kbgxi1060gwgz

pretoriano80, 2022-08-28:
However,i think there's more to it than just the PWM freq setting as i've tried different combinations and nothing worked as it does with the Tuya app.

DSchndr, 2022-08-28:
Best would be to measure with an oscilloscope what the tuya fw does, but beware, the power supply is not isolated!
I kinda dont want to take apart this lamp again since putting it together is like a puzzle 😂
I do have the firmware dump tho, just no tuya modules to flash it onto :/

pretoriano80, 2022-08-28:
The one i have it's pretty hard to open but eventually i will try again .
I wonder if it's possible to flash back the stock firmare (with or without disassemble ).

pretoriano80, 2022-09-03:
@openshwprojects any thoughts on this?
I couldn't find any specific info about KP18055ESPA IC in Tuya documentation ,so i don't think they used a specific implementation for this PWM IC.

Maybe OpenBeken PWM driver is missing something for this to work properly ?

pretoriano80, 2022-09-04:
Nevermind,i had a few esp-12f around so i decided to cut it open (led base was glued so it was impossible to remove ) and replace the WB2L module with esp-12f ,it wasn't easy but it was fun and i managed to close and glue the bulb back.

As far as the functionality ,with Tasmota it works flawlessly ,so it's not affected by this issue ,brightness dimming is linear for both RGB and CC/CW modes .

I do have another (identical ) one which i will have to decide what to do with it . xD

DSchndr, 2022-09-05:
@pretoriano80 Could you test this firmware on the bulb? (Just do ota upload) Should fix the issue :)
Thanks for nudging me into looking closer at the pwm code xD (tl;dr: beken sdk is shit)
@openshwprojects fixed pwm code is in the archive

[OpenBK7231T_App_DSch_PWMFIX.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/9491029/OpenBK7231T_App_DSch_PWMFIX.zip)

DSchndr, 2022-09-05:
Increased PWM range (1-1000 instead of 1-100) would be nice for finer control in the lower dimming region
Or an option to make it non-linear in the lower region

pretoriano80, 2022-09-05:
@DSchndr  i will test in a few days,when i get home .
I will have to "unlock" the second bulb first.,the first one is tasmotized xD.

DSchndr, 2022-09-05:
Suka Blyat, there is another bug in 
https://github.com/openshwprojects/OpenBK7231T_App/blob/3b0fbc768b6fade279be9e22632838b339cacfb8/src/cmnds/cmd_newLEDDriver.c#L173

Should be fixed in this ota binary 

[OpenBK7231T_App_DSch_PWMFIX_2.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/9491453/OpenBK7231T_App_DSch_PWMFIX_2.zip)

pretoriano80, 2022-09-05:
@DSchndr so your device is working fine now?

DSchndr, 2022-09-05:
Yes, cw/ww + dimming in Homeassistant working properly and the lamp pulls its 14w with my last binary, if you could test I would make a PR or openshwprojects could add the changes in :P

pretoriano80, 2022-09-06:
@DSchndr just made a quick test abd now it works good.
Only thing is that it seems that peak brightness is reached before 100 (on the 0-10 scale),but i will have to test with a power meter or something.
However,great work,the bulb is perfectly usable now.

pretoriano80, 2022-09-06:
Ok,finished the mqtt configuration and tested with Home Assistant.I've tested with a power monitoring plug and the brightness steps are perfect ,50% = 5w ,70%= 7w and 100% = 10w (this is a 10W bulb).

So as far as i can say (i can only test on this device) your PR should be good to go.

pretoriano80, 2022-09-11:
@DSchndr  can you provide a diff/patch file with your changes ?Thx !

DSchndr, 2022-09-11:
@pretoriano80 https://github.com/openshwprojects/OpenBK7231T_App/pull/194

closed 2022-09-11 by DSchndr