openshwprojects/OpenBK7231T_App #153: BK7231T e14 Bulb Pin Mapping

bonuzzz, 2022-08-09:
I bought tuya e14 bulbs on aliexpress some time ago. I think there more noname than branded one.  I flashed them with your firmware and want to publish working pins:

pin6 pwm 4
pin7 pwm 3
pin8 pwm 5
pin24 pwm 2
pin26 pwm 1

[image]
[image]
[image]
[image]

openshwprojects, 2022-08-09:
Thanks, but the better place for that would be the Teardowns section here:
https://www.elektroda.com/rtvforum/topic3911041.html
with an information where that bulb was bought and how was flashed

bonuzzz, 2022-08-18:
@ardenking yes, I have

```
mqtt:
  light:
    - name: "Left Lamp1"
      state_topic: "left_lamp1/led_enableAll/get"
      command_topic: "cmnd/left_lamp1/led_enableAll"
      brightness_state_topic: "left_lamp1/led_dimmer/get"
      brightness_command_topic: "cmnd/left_lamp1/led_dimmer"
      rgb_state_topic: "left_lamp1/led_basecolor_rgb/get"
      rgb_command_topic: "cmnd/left_lamp1/led_basecolor_rgb"
      color_temp_command_topic: "cmnd/left_lamp1/led_temperature"
      color_temp_state_topic: "left_lamp1/led_temperature/get"
      rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
      rgb_value_template: "{{ value[0:2]|int(base=16) }},{{ value[2:4]|int(base=16) }},{{ value[4:6]|int(base=16) }}"
      qos: 0
      payload_on: "1"
      payload_off: "0"
      optimistic: false
      brightness_scale: 100
      brightness_value_template: "{{value}}"
```

ardenking, 2022-08-19:
thank you works perfectly

ferbulous, 2022-09-08:
Hi, may i know how you dismantled the bulb and assemble it back?
Especially with the wires connected to the bottom base

bonuzzz, 2022-09-08:
@ferbulous I'll write about it later. But I flashed remaining lamps with tuya-cloudclutter with flsnt br30 rgbwct profile without dismantle