openshwprojects/OpenBK7231T_App #154: NTP Driver fails to get time

iprak, 2022-08-10:
I have not had success getting NTP driver to work. I manually started it using the command `startDriver NTP` but continue to see this in the log

```
Info:MAIN:Time 547, free 103352, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 548, free 103584, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 549, free 103584, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 550, free 103352, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38

Info:GEN:dhcp=0 ip=192.168.1.212 gate=192.168.1.2 mask=255.255.255.0 mac=50:8a:06:cf:e5:52 

Info:MAIN:Time 551, free 103584, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 552, free 103584, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 553, free 103344, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 554, free 103584, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 555, free 103584, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 556, free 102976, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 3/38

Info:NTP:NTP_CheckForReceive: Error while receiving server's msg

Info:MAIN:Time 557, free 103216, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 3/38

Info:NTP:NTP_CheckForReceive: Error while receiving server's msg

Info:MAIN:Time 558, free 103216, MQTT 1, bWifi 1, secondsWithNoPing -1, socks 3/38

Info:MAIN:Forced channel publish! Publishing val 1 with 1 
```

I have verified that the target `217.147.223.78` is accessible.
```
Pinging 217.147.223.78 with 32 bytes of data:
Reply from 217.147.223.78: bytes=32 time=142ms TTL=41
Reply from 217.147.223.78: bytes=32 time=141ms TTL=41
Reply from 217.147.223.78: bytes=32 time=142ms TTL=41
```

openshwprojects, 2022-08-11:
BK7231T or N?

It works for me, at least on that T build:
[image]

iprak, 2022-08-12:
That was my bad. I forgot that my devices are firewalled. Do you think it would be worth allowing custom NTP server via a command?

closed 2022-08-12 by iprak