openshwprojects/OpenBK7231T_App #155: Mapping CW and WW as a single

fergbrain, 2022-08-13:
Feit Electric Wi-Fi 4’ Smart LED High Output 10,000 Lumen Shop Light from Costco (Model SHOP/4/HO/CCT/AG). It has a WB2L (BK7231T) chip and I've successfully been able to flash with v1.12.26.

P7 is a PWM that controls color temperature
P8 is a PWM that controls dimming
There's another I/O on P15, but I haven't been able to figure that out yet, but it seems to need to a dInput, otherwise the light doesn't work.

On the main (index) page there's are two sliders:
* LED Dimmer/Brightness 
* LED Temperature Slider

However, they don't work. Is that a known issue or do I need to fix something in my configuration?

I can use the web app to successfully control the light, but it shows as:
* one slider for "CW", but is actually for color temperature
* another slider for "WW", but is actually for dimming

Is there a way to configure that to be correct?

openshwprojects, 2022-08-14:
Oh, thanks for reporting, there indeed might be an issue of that.

I will try to update LED driver today, but could you please check if it works if you set 3 extra dummy PWM pins to channels 0 1 2 and set CW and WW PWMs to channel 3 and 4? So device "thinks" it's RGBCW?

Anyway, I will look into doing a proper fix today, but I would need you to help with testing, because I don't have CW lights at hand.

fergbrain, 2022-08-14:
That sort of works.

**Configuration:**
P6: PWM (0) -- Dummy
P7: PWM (3)
P8: PWM (4)
P9: PWM (1) -- Dummy
P24: PWM (2) -- Dummy

**Results**
The "LED Dimmer/Brightness" changes both ch 3 and ch 4 together.

The "LED Temperature Slider" changes both channel 3 and 4 inversely:
* When set to "Cold": Ch 3 is 100 and Ch 4 is 0
* When set in middle: Ch 3 is 50 and Ch 4 is 50
* When set to "Warm": Ch 3 is 0 and Ch 4 is 100

The issue is this light uses:
P7 (3) _just_ for color temperature: 0 is cold and 100 is warm
P8 (4) _just_ for dimming control: 0 is off and 100 full brightness

Happy to help and test! I'm still trying to read through and understand the code base, so hopefully I can be more help in the future!

openshwprojects, 2022-08-14:
I just realized my previous statement was incorrect.

If you have:
- one PWM for dimming
- second PWM for Warm/Cold

then you could just set two PWMs in pins (and nothing more) and then go to Config/General and set:
**Flag 3 -[LED][Debug] Show raw PWM controller on WWW index instead of new LED RGB/CW/etc picker**

this is not perfect and does not provide On/Off, but should settle for at least today. Then I will update LED driver to support your case.

fergbrain, 2022-08-14:
That works for now.

What other information can I provide that would be helpful?

I have photos the PCB. I also have an oscilloscope and other tools. I have two more of these lights I have not yet modified, so I could probably dump the existing firmware if that would be helpful.

openshwprojects, 2022-08-16:
can you check latest update?
https://github.com/openshwprojects/OpenBK7231T_App/commit/55c308813cb1a5a49a859bf3abe90c89ffb90836
Set only two PWMs in config.
[image]
Set this flag:
[image]
Do I need to inverse one of the channels? I would guess so, but which?

Is the basic idea working?

fergbrain, 2022-08-17:
Hey! It works!

Cold/Warm needs to be flipped, but other than that it works!

MQTT (led_temperature and led_dimmer) also works as well, though the temperature is also flipped (which I would expect).

openshwprojects, 2022-08-17:
I think the simplest solution would be to add a PWM_n pin which is like PWM but flipped... maybe I will also flip it in code, but do we know if the Cold/Warm order is standarized among manufacturers?

I will flip it for you in code and also add PWM_n when I come back home later today

fergbrain, 2022-08-17:
I would think that 0 would be Cold and 100 would be Warm, but obviously this light is the inverse of that. I haven't seen CW/WW implemented this way on other lights, so I don't know if this is the standard way to do it or not...or if there even is a standard.

If you're able to add PWM_n, that works for me too. Either way.

openshwprojects, 2022-08-17:
I didn't flip it yet, but does it work correct if you set pin role to IOR_PWM_n ?

fergbrain, 2022-08-17:
Yes, it works correctly when I set pin role to PWM_n.

openshwprojects, 2022-08-17:
That's great news. Just for the sake of information, do you happen to have some photos from device inside?
You can post a short teardown/info post here if you have some free time: https://www.elektroda.com/rtvforum/forum507.html

Is there anything else you need to be added?

fergbrain, 2022-08-17:
Yes, I have lots of photos and would love to do a teardown post! I'll work on that later today :)

That's it! Works perfectly and I already have it integrated into my Home Assistant! Thank you so much!

closed 2022-08-17 by fergbrain

openshwprojects, 2022-08-17:
Thanks. 

This device is especially interesting for me because I have never seen previously such a strange PWM cold/warm control. I wonder which chip it uses for LED control

fergbrain, 2022-08-18:
I'm still working on my write-up. But I wanted to let you know that I did find the chip: It's a [Bright Power Semiconducter 
BP5929](http://bpsemi.com/cn/product_result.php?id=572). The datasheet is in Chinese but this graphic aligns with what it appears to be doing:
[image]

fergbrain, 2022-08-23:
Write up completed:

https://www.elektroda.com/rtvforum/topic3914634.html

And also on my blog: https://andrewferguson.net/2022/08/23/feit-shop-4-ho-cct-ag-teardown/