openshwprojects/OpenBK7231T_App #156: ELIVCO LSPA9 BL0937 reports power, but not voltage or current

analystcmyk, 2022-08-14:
Firmware : Build on Aug 5 2022 08:18:26 version 1.12.15
Chipset BK7231T
Brand : ELIVCO LSPA9
Power metering chipset : BL0937
Module  : WB2S

Config : 
P6 = BL0937CF1
P7 = BL0937CF
P24= BL0937SEL

Power is reported correctly in debug log and MQTT, but no values for V and I are reported. 

Any suggestion for further debugging is appreciated. 

Original issue : 
https://github.com/xeniter/WIFIPO120FWT/issues/1

bonuzzz, 2022-08-14:
I have BL0937CF1 on Pin8

analystcmyk, 2022-08-15:
@bonuzzz  Thank you for the tip, I changed it to p8 but it did not change , still no voltage/current.

analystcmyk, 2022-10-11:
Yes, this is solved, thank you.

I have tested this device also to check if it supports reverse power ( i.e. when connected to grid tied inverter).

Indeed it will report reverse power the same way as for forward power.

closed 2022-10-11 by analystcmyk